dwhinham/mt32-pi #1: usb connection needs clarification

likov, 2017-05-28:
hello there! could you please provide more details - how to connect an RPi to a PC? is there a USB-USB midi cable? or need I a pair of USB->MIDI and MIDI->USB cables? or a way to connect USB->MIDI to GPIO?

jeloneal, 2020-07-02:
Hi! I'm also interested in this topic. I was searching for a replacement for a real HW MT-32 to attach to my retro computers (mainly Ataris) that have MIDI interfaces. Is your solution intended to be able to act like this? And if so, how?

dwhinham, 2020-07-02:
Hey!

Yes - the idea is that your Pi becomes a drop-in replacement for an MT-32.

So I've sortof "revived" this project - before, it was a minimal, tiny Linux image that booted directly to the MUNT MT-32 emulator, but now, it's a bare metal project. This means that there's no operating system, it's smaller, and it starts in about 4 seconds from cold boot!

Please note that it's in very early stages, there are audio quality issues (especially on Pi 3).

I'll be writing some proper documentation soon, but essentially, it goes like this:

- Get the latest build by going to Actions -->Click top-most commit in list --> Click `sdcard` in the artifacts list.
- Extract contents to a blank FAT32-formatted SD card.
- Add `MT32_CONTROL.ROM` and `MT32_PCM.ROM` to the root of the SD card - you have to provide these for copyright reasons.
- Plug a USB MIDI interface into the Pi, and connect some speakers to the headphone jack.
- Connect your vintage PC's MIDI IN to the Pi's MIDI OUT and vice versa.

The USB MIDI interface connected to the Pi can be any standard class-compliant USB MIDI interface. If it works on Windows or Linux without any drivers, there's a high chance of it working.

Examples:
```
[ Pi ] --> [ USB MIDI ] <===> [ USB MIDI ] <-- [ Modern PC ]
[ Pi ] --> [ USB MIDI ] <===> [ Gameport MIDI cable ] <-- [ Vintage PC ]
[ Pi ] --> [ USB MIDI ] <===> [ Synthesizer keyboard or controller ]
```

nswaldman, 2020-07-03:
I'm currently using an M-Audio Midisport 1x1 connected to a Roland UM-One, works fine.

jeloneal, 2020-07-04:
Very cool stuff. I followed your instructions and connected the whole thing (RPi 3B plus DigitalLife MIDI to USB) to my old ATARI Mega STE. The music works, but it seems as if there are slow downs or timing errors. Could it be that RPi's power management or dynamic clock changes are interfering with munt's timing? When testing against Munt on my windows pc that didn't occur and I remember reading about this cpu clock changes affecting munt on RPi: https://retropie.org.uk/forum/topic/12549/tutorial-installing-munt-mt-32-emulation-on-rpi-3 (cpu-governor stuff under numbers 12 and 13 there).
Does the bare metal approach allow for adjustments of that stuff?

dwhinham, 2020-07-04:
Thanks for testing!

When did you download the package? A couple of days ago I received a Pi 3 B+ and made some fixes to eliminate the stuttering. Basically the Pi was running at 700Mhz on startup and needed some additional code to bring it up to full speed - the stuttering was caused by it not running fast enough to handle more than a certain number of voices simultaneously playing. I pushed this new code yesterday. 🙂

Could you redownload the latest code from Actions and make another test? You can just choose the "kernels" package and replace those files on the SD card; all the other files stay the same.

Cheers!

jeloneal, 2020-07-04:
I just downloaded and tested today. So I already had the most recent build. But I will download again just to make sure.

dwhinham, 2020-07-04:
By the way, was it distortion in the audio (crackling/popping) or more like MIDI notes being played 'late'? The latter is another problem that's on the TODO list.

jeloneal, 2020-07-04:
It was no crackling but rather notes missing or being played late.

dwhinham, 2020-07-04:
Okay, there's probably stuff to be done there - would you mind opening another issue about this topic?

If you can provide details of the MIDI file(s) you're playing, how it should sound versus how it actually sounds, and maybe a way I can reproduce the issue (e.g. can I play something from a PC or Atari ST emulator and hear the problem?), this will help me a lot.

Thanks!

jeloneal, 2020-07-04:
Just a last addition. I downloaded again and did a fresh card. On Pi3 it's still having the issues described. Then I tried my Pi4 just out of curiosity et voilá: Pi4 seems to work flawlessly.

Will do further testing and let you know.

P.S. Perhaps you can add your desctription from the earlier post as readme file?

dwhinham, 2020-07-04:
Thanks - that's good to know.

Audio quality: yes, that's already known - I don't thank that's going to get any better. The Pi's headphone jack is known to be very rudimentary (it's not a true DAC).

The best way to use this project will be with a DAC - here's a video showing the audio quality from the DAC I've been working with: https://www.youtube.com/watch?v=Q0_ay0dCnRU

I'm working on updating the README now. 😃

commit f894611e (2020-07-04): Update README.md

Closes #1.
closed 2020-07-04 by dwhinham