dwhinham/mt32-pi #2: It would be interesting to see this implemented in bare-metal (no OS, or very, very minimal OS)

jan-kleks, 2018-02-08:
This way you could get even faster boot times, and manage the Pi resources even more effectively. Check out, for instance, this bare-metal Pi stuff:

[Raspberry Pi plays MIDI without an operating system](https://hackaday.com/2013/01/22/raspberry-pi-plays-midi-without-an-operating-system/)

[Raspberry Pi Emulates an Amiga 500 Floppy Drive](https://www.reddit.com/r/fpgagaming/comments/7vrytw/raspberry_pi_emulates_an_amiga_500_floppy_drive/)

[Raspberry Pi Co Processor that directly attaches to the Beeb](https://github.com/hoglet67/PiTubeDirect)

[Raspberry Pi bare metal forum section](https://www.raspberrypi.org/forums/viewforum.php?f=72&sid=f0b46ef56a2efd34a34a125d12be21ba)

dwhinham, 2018-02-09:
This would be a really nice idea and probably the best way to do it - thankyou for the links, I wasn't aware of how much progress had been made with baremetal Raspberry Pi development.

The Circle library (https://github.com/rsta2/circle) looks like it would be ideal for this, as it has MIDI, filesystem (to read ROMs?) and audio support. It would be a case of attempting to port libmt32emu to this environment.

If I get some free time in between University I might look into how feasible this is.

Cheers!

jan-kleks, 2018-02-24:
@dwhinham Maybe you could even try to add a screen and buttons to make it even more like a real device? Keep up good work!

dwhinham, 2020-07-04:
I'm going to close this issue, because (ahem, 2 years later...) it's now happening! 😄 
Screen and buttons are definitely on the TODO list.

Thanks for the inspiration!

closed 2020-07-04 by dwhinham