dwhinham/mt32-pi #4: A better project name?

dwhinham, 2020-07-04:
The current project name isn't very exciting; it's just what I typed when starting the repository!

Can you help come up with something better? 😃

stt, 2020-08-05:
(Don't have a naming suggestion exactly, just ruminating on naming practices in general..)

Being functionally descriptive mt32-pi is pretty good. Where naming issues arise is when there's multiple projects doing essentially the same thing and use similar names or when project changes and the functional name becomes inaccurate. Maybe this project starts emulating something else than mt32 or runtime platforms other than pi.

Some approaches people take to avoid those is to either be more abstract (soundemu-embed might be a bad name but it'd cover more ground), or go for petnames.

I've been entertaining a fantasy that github could start supporting optional part in reponame, kinda like gmail addresses can have `name+somelabel@gmail.com` where somelabel doesn't affect where it's delivered but can be used for context.
In that scheme if this project would e.g. adopt name Rollo the reponame could be Rollo=mt32-pi.. and if the tech changes then Rollo=mt32-sbc would open the same project.

dwhinham, 2020-08-10:
Thankyou for your rumination and suggestions! 😃 

There's been some discussion about expanding the project to other synths (#18, #19), and so your points about a more abstract name are very relevant.

I encourage anyone to just drop ideas in here, no matter how silly; it may inspire us.

I'll start: If we manage to get a few more synth engines included, maybe a petname like Swiss Army Baremetal Synthesizer / SwABS could work. The logo could be a Swiss army knife, but with all sorts of waveforms coming out of it. 😄

thorr2, 2020-09-20:
I like the current name, but if you want to change it after implementing more midi modules, another idea is: 
BaMMMM!:  B(a)re Metal Multiple Midi Modules

closed 2021-04-30 by dwhinham