dwhinham/mt32-pi #7: LCD/OLED screen support

dwhinham, 2020-07-04:
One of the nice things about the desktop version of MUNT is that it simulates the LCD screen of the real device.

It would be nice to be able to hook up a cheap character LCD to the Raspberry Pi so that status information is visible along with the MT-32 SysEx messages that some games like to display on the screen.

I'm currently implementing this for a 4x20 HD44780-style screen, and will probably add support for 2 and 1-line screens too.

Demo of some work-in-progress code can be seen here:
https://www.youtube.com/watch?v=Q0_ay0dCnRU

What screens would you like to see working?

nswaldman, 2020-07-06:
It would be very nice to see support for 128x32 OLED displays -the ubiquitous 0.91" version is very popular with Gotek and Arduino tinkerers and should be rattling around in a lot of drawers.

SWLee0129, 2020-07-07:
I like this lcd more because it looks like MT-32.  Hope all the other functions will work just like the real device.

commit 6d9c3906 (2020-08-03): Add LCD/OLED support and documentation

Initial support for HD44780 (4-bit direct and I2C) and SSD1306 OLEDs.

Closes #7.
commit f938f613 (2020-08-03): Add LCD/OLED support and documentation

Initial support for HD44780 (4-bit direct and I2C) and SSD1306 OLEDs.

Closes #7.
commit 28861625 (2020-08-03): Add LCD/OLED support and documentation

Initial support for HD44780 (4-bit direct and I2C) and SSD1306 OLEDs.

Closes #7.
closed 2020-08-03 by dwhinham