dwhinham/mt32-pi #13: Rainbow screen

squeeeb, 2020-07-18:
I'm getting a rainbow screen on boot with the newest release.  Using a Raspberry pi 3 Model B.  I believe I've ruled out the hardware because:

- this is occurring on two different Sandisk SD cards (formatted with the official SD Card Utility)

- I'm able to boot/run Raspberry Pi OS (Minimal image based on Debian Buster) with no issue. I also ran the 'full-upgrade' while this was booted

- I can query the i2s DAC once in Pi OS (returns 4d)

The only thing I haven't done yet is hook up my Roland UM-One via USB - would this prevent the kernel from booting?

Any help would be appreciated, this looks like a great project!

dwhinham, 2020-07-18:
Hey :)

This is normal; there's no video output - the kernel runs 'headless' unless it's compiled in debug mode.
If you connect a MIDI device and play some notes, you should see the LED blink when a message is received.

I should probably clarify this in the docs.

squeeeb, 2020-07-18:
Oh...well then 😅

Thanks for the quick response!

dwhinham, 2020-07-18:
No worries! 😄 Enjoy!

closed 2020-07-18 by dwhinham