dwhinham/mt32-pi #14: is possible use an Adafruit DAC?

DSkywalk, 2020-07-30:
Hello! 
Is it possible to use my Adafruit UDA1334A in this project?
How should I set it up?

Documentation for raspbian:
https://learn.adafruit.com/adafruit-i2s-stereo-decoder-uda1334a/raspberry-pi-test

My settings on volumio:
[image]

I'd love to try this project at home.
Thank you very much.

dwhinham, 2020-07-31:
Hi there! 😃 

Don't worry about `i2cdetect` not finding anything - this DAC is not controlled by software and so we can ignore the I2C side.
It *should* be enough to set `output_device = i2s` in the config, and ignore the `i2c_dac_xxx` options.

Let me know if it works and we can add your DAC to the compatibility list!

You can test by playing MIDI notes on channel 2 or above, or by hooking it up to a PC running ScummVM - you should see the Pi's activity LED blinking when it's receiving data.

dwhinham, 2020-08-10:
Hey there, just wondering if you'd managed to get your DAC working? 😃 

If so we can close the issue. Thanks!

DSkywalk, 2020-08-10:
Sorry, my first idea was to directly use a serial port of an FPGA, but we still don't have the correct rate configuration. Anyway, some sound should come out, because if I use PWM, random sounds are heard. So I thought about building the whole circuit to have better tests, but I still have to get the parts.

If you want I can close it, although as I say I haven't been able to hear anything with the I2S connected :disappointed:

dwhinham, 2020-08-10:
No need to close the issue, just checking in! 😃 

Yeah, those logs are showing garbage MIDI data and Active Sense is getting confused.
A valid note-on should look like `91 0C 7F`.

What's the baud rate of the MiSTer? `mt32-pi`'s serial baud rate is set to standard MIDI 31250bps here: https://github.com/dwhinham/mt32-pi/blob/0147c2c61e907e59f2ca8431cc9da533e98e2584/src/kernel.cpp#L100
You could try 1152000 or some other rate to match the hardware - if it works, I'll create a config option.

Your DAC should definitely work as I'm using another board with UDA1334A and it works great as long as I set the config to use i2s!

I do have some code somewhere that I used to self-play a test pattern for my friend who had no MIDI but wanted to check his DAC. I'll see if I can find it!

yavimaya, 2020-08-11:
> 
> 
> Hey there, just wondering if you'd managed to get your DAC working? 😃
> 
> If so we can close the issue. Thanks!

I CONFIRM it works perfect with both ES9018K2M and UDA1334 dacs :D

dwhinham, 2020-08-31:
Just going to go ahead and close this as the original question has been answered. Thanks!

closed 2020-08-31 by dwhinham