dwhinham/mt32-pi #15: Alter (O)LED layout

nswaldman, 2020-08-03:
During testing of the generic (128x32) OLED display it struck me that the layout felt off -the original 20x1 character output is on the first row and the volume bars on the second. Visually it would make sense to swap these around, combined with blanking out the appropriate channel completely rather than inverting it -just like on the original MT-32.

I feel it would be a good idea to explore the rest of the multiline displays for the most useful output too, but I guess that's to be explored in discrete tickets :)

nswaldman, 2020-08-03:
Blanking out the channel already happens in 0.3 -no flies on mt32-pi 👍

dwhinham, 2020-08-03:
Thanks!

Actually on looking closer, I forgot to do the blanking-out on the 'R' channel, so that's now fixed!

I implemented your suggestion and I think it does look better:
[image]

If you think this is good I'll probably just merge it, unless you have any other layout suggestions! 😃

nswaldman, 2020-08-04:
It would be great if the volume bars were aligned with its respective channel?

dwhinham, 2020-08-04:
Maybe you can help me design this better. The reason it's like this is because it's a compromise: there are 9 bars (for the 8 tone parts and +1 rhythm part), and the bars are evenly spaced and have widths to use up all the avaialbe space, but the bottom row is closely simulating the full 20 characters of the original MT32 display (which goes 1 2 3 4 5 R, where R is actually 9).

Here's how it's mapped (at least, before I swapped the rows):
[image]

I could remove the last 4 bars and align the first 5 + R to the indicators, and do something else with the remaining space?

nswaldman, 2020-08-04:
I'll have a stab at this, thanks for the example.

nswaldman, 2020-08-17:
...and I'm fresh out of ideas :) I kept mentally trying to map a bar with a channel digit and that's _not_ how it works.

commit 86be95e1 (2020-10-03): Move level meters to upper row of LCD

Closes #15.
closed 2020-10-04 by dwhinham