dwhinham/mt32-pi #17: "[lcd]" section missing in mt_32.cfg for release 0.3

jsmolina, 2020-08-08:
Hi,

I've been looking for the option to enable the MIDI pi-hat LCD, until I realised that repo had this 'sdcard' folder. It should be included on release.

Very nice job!!!

Thanks!

dwhinham, 2020-08-08:
Thanks for reporting this!

This unintentional of course - I think the caching system in GitHub Actions (which I use to speed up the build) is overwriting the latest config with an old version.

I'll take care of this and prepare a new release very shortly. 😃

closed 2020-08-08 by dwhinham

dwhinham, 2020-08-08:
[v0.3.1](https://github.com/dwhinham/mt32-pi/releases/tag/v0.3.1) is now available with the missing config section included and some other minor fixes.

Thanks again! 😃

commit 9ba8a2b0 (2020-08-08): CI: Be explicit about boot files to cache

This should prevent old versions of the config files being cached.

Fixes #17.