dwhinham/mt32-pi #19: feature suggestions from WavePi dev

t9999clint, 2020-08-09:
Yo, I like your project. It's almost overtaken mine in a short amount of time.
My project is right here. Sadly I haven't been able to work on it as much as I'd like in the last year or so. https://github.com/t9999clint/WavePi

I have some experience with FluidSynth if you'd like, but I'd warn against it as it has quite a few bugs that I've never been able to work out. It seems like it's not handling modulation and pitch bend commands correctly and it glitches out after about 20 minutes.
I'm using BassMidi as a replacement right now, but it's closed source so I haven't been able to share it.

So here's what I'd suggest you add in order to have it completely overtake my project (the software side) in terms of functionality...

1. Add some sysex commands for switching synth modes (Between Fluidsynth soundfonts, MT-32, CM-32, etc...)
2. Add the ability to use midi over serial (not really needed but it's nice)
3. Add Fluidsynth or some other opensource alternative so it can load soundfonts

Feel free to contact me for any questions

dwhinham, 2020-08-09:
Hi there!

Thankyou for the suggestions, I appreciate what you've done with WavePi!

I haven't yet done any tests of FluidSynth with a standard Pi Linux setup just yet, so I'll look out for some the issues you've described.

It'd probably take me some time to get FluidSynth ported to bare metal as it has a dependency on glib which we don't have, but I'll be giving it a shot.

Some of these ideas are being discussed over in #18 too.

Have you had any experience with/had any thoughts on Timidity++ or WildMIDI? I thought it might be worth ruling them out before diving into a FluidSynth port attempt.

Thanks! :smiley:

t9999clint, 2020-08-10:
I was never able to get Timidity to compile (although I didn't try very hard) and I've never herd of WildMIDI yet,
I'll give those a shot and let you know.

t9999clint, 2020-08-10:
Looks like WILDMidi doesn't support sf2 files so it's not a suitible replacement for FluidSynth, but it DOES do GUS patches which I think is really good if you want to add Gravid Ultrasound support to your project. I'd probably include it in addition to FluidSynth.
I'm still testing Timidity I'll let you know what I find.

calvinmorrow, 2020-10-18:
Reading through this issue got me curious, so I went looking around.  Looks like there have been at least two efforts to remove glib from FluidSynth.

There's a new-ish one I'm documenting here in case its of use later on: https://github.com/chirs241097/fluidsynth-sans-glib/commit/04e566f776b08d7ea4e69690d8bc0578833885e4

dwhinham, 2020-10-18:
It's really funny you should mention this.

As of literally about 15 minutes ago, I got the first sounds out of FluidSynth from bare metal. 🙂

I looked at some of the forks without glib and also FluidSynth Lite, but I wanted to try my own approach of just patching mainline FluidSynth so that I can be sure that I can stay in sync without depending on a fork.

That approach has succeeded. The next version of `mt32-pi` will be a very exciting release! 🙂

dwhinham, 2020-10-18:
I just Tweeted a short demo for those interested in FluidSynth 🙂

https://twitter.com/_d0pefish_/status/1317914900866490368

dwhinham, 2020-11-22:
I just released [v0.7.0](https://github.com/dwhinham/mt32-pi/releases/tag/v0.7.0) with FluidSynth support, taking care of the final suggestion on the list, so I think we can close this. 🙂

closed 2020-11-22 by dwhinham