dwhinham/mt32-pi #20: Wrong instruments

loskaa, 2020-08-10:
I am currently having small issue with MT32-pi. It’s playing music but all the instruments are wrong and they are different every time when I boot the raspberry. The MT32-pi is connected to old dos pc using game port midi cable and m-audio midisport uno. I tried windows 10 laptop with munt instead MT32-pi and it is playing instruments correctly.

I am probably missing something pretty obvious here but cannot figure out what :)

dwhinham, 2020-08-10:
Hi!

That's interesting - which game(s) are you trying, and are they definitely set up in MT-32 mode (and not, e.g. General MIDI mode)?

Most games will send a burst of SysEx to the MT-32 when in the correct mode to set up the instrument assignments before playing the actual music.

loskaa, 2020-08-10:
I have tried Monkey Island 1/2 (using r parameter), Space Quest 3/4 and The Legend of Kyrandia. Softmpu  is running and the lcd display connected to pi is showing “Lucasarts presents” and other messages. 

MT32-pi is running before loading any games. 

Don’t worry. I am fairly certain it’s something obvious on my side 😄

dwhinham, 2020-08-10:
Ahhh, now SoftMPU isn't something I've actually tried yet on my 486 as I use an intelligent mode MIDI interface, so that could be the difference.

I noticed this from the [troubleshooting section of SoftMPU's docs](http://bjt42.github.io/softmpu/):
> Problem: My Rev.0 MT-32 displays “EXC. BUFFER OVERFLOW”. Music playback sounds incorrect.
Solution: Run SoftMPU with the /DELAYSYSEX switch.

Could you give that a try?

If there's no joy, I'll have to dig out a gameport cable and try to reproduce this on my 486 with AWE 32.
What's the CPU in your DOS PC by the way?

Thanks 😄

loskaa, 2020-08-10:
No difference with /delaysysex switch. Should the games which doesn’t require intelligent mode work without midi interface card or softmpu with MT32-pi? When softmpu isn’t running, all the games I’ve tried still have wrong instruments. 

I have 233 mmx in this pc.

dwhinham, 2020-08-10:
Honestly, I'm not sure! I haven't really dealt with these kinds of compatibility issues before - sorry! 😞 

What kind of sound card do you use? I know some certain version of Sound Blaster have MIDI bugs...
Did you also say that connecting this PC to your modern PC running MUNT using the same USB cables works fine?

The LCD messages working is promising, at least - that means some SysEx is getting through fine.
I should test using a USB interface on the Raspberry Pi again because it's been a while since I last tried (I now use GPIO and it's entirely possible that I broke something for USB without noticing!).

loskaa, 2020-08-10:
The soundcard is Yamaha audician 32. Yes, if I change pi to pc with munt (same cable and same usb midi interface) all sound just as they should.

dwhinham, 2020-08-10:
I just tested SoftMPU (no special switches, just `/MPU:330`) with my Sound Blaster AWE32's joystick port, going into my Pi both via GPIO and via USB interface (M-Audio USB Uno) - everything working fine with Monkey Island 1 and 2 with the `r` option.

I'm running out of ideas - something's probably getting corrupted along the way.

What kind of Raspberry Pi are you using, and what kind of power supply are you using to power it?
If it's too weak that could explain bad behaviour.

Do you have any other sound cards you could try in the Pentium?

When I get some time I could do a special build for you that logs all incoming MIDI data to a file on the SD card. That way we could try to get a better idea of what's happening.

loskaa, 2020-08-10:
It’s raspberry pi 3b and the power supply is 2.5A model with raspberry logo on it. I tried another power supply, another pi and few different sd cards to rule them out but the results are the same. I did also try another pc with sound blaster 32 but also the same problem. 

I had to do an adapter to connect two male midi cables (from game port cable to uno) which you can call a hack job but it’s working with the windows laptop with munt. 

But thank you for confirming It’s working perfectly with similar setup. I’ll try to find what stupid mistake I have done 😄

dwhinham, 2020-08-10:
The adapter/coupler should be fine, really - I know someone else using the same idea - but who knows, there could be some electrical incompatibility.

I've done a special build here: https://github.com/dwhinham/mt32-pi/actions/runs/203033264

This build will append everything it receives via midi to a file called `midi-log.txt` on the SD card - it will look a bit like this:
```
93 34 24
91 28 30
93 3B 20
93 40 1B
93 42 1E
94 58 34
94 53 28
94 4F 30
94 4E 30
94 4A 12
...
```
When you get some time:

- Download the `kernels` artifact from the link above, and replace the `kernel*.img` files on your SD card with the ones from that archive.
- Boot your Pi, and let the Monkey Island 1 title music play through until Guybrush meets the guy next to the fire.
- Quit the game, turn off your Pi, retrieve the `midi-log.txt` from the SD card and attach it here.

I can then compare it to what it should contain, and we can see if there's any pattern to what's getting lost.

Cheers!

loskaa, 2020-08-10:
[midi-log.txt](https://github.com/dwhinham/mt32-pi/files/5053268/midi-log.txt)

Ok, here we go.

dwhinham, 2020-08-10:
Okay, here's the start of a diff, with my "good" log on the left and yours on the right. I noticed yours includes the SoftMPU LCD message SysEx, which looks good (I can "replay" that data on mine and I see the message).

[image]

Your log is missing so much data - there are huge chunks of CC messages and program changes missing, and also it looks like a lot of the packets that do get through are corrupted/have wrong values.

I think there may be some kind of electrical problem or incompatibility corrupting the data.

A few more ideas:
- Have you tried other USB ports on the Pi?
- Are you using a DAC?
- Have you changed any `mt32-pi.cfg` settings related to audio buffer sizes?
- You mentioned using an LCD - does anything change if you disconnect it?
- Can you try the MIDI interface connected to a powered USB hub and then to the Pi?
- Have you tried running Linux (Raspberry Pi OS) on the Pi, installing MUNT and using it this way?
  - If the problem still happens here, we have some kind of incompatibility with with the combination of this USB interface, your Pi, and your vintage PC.
  - If the problem goes away, it might be an issue with the Circle (bare metal) USB MIDI driver.
- Do you have any other way of 'coupling' the PC to the USB MIDI interface? You could connect the OUT of the vintage PC to the IN of a MIDI keyboard or synth for example, and then connect the Pi's USB interface to the THRU on the synth.
- Could you get another USB interface to try?

What's strange is that I have an Uno too that works fine, it looks just like this:
[image]

Mine is very old, have had it since the early 2000s, but reliable - I wonder if there have been any revisions to it that affected compatibility.

Sorry you're having trouble - I really want to help you get this resolved! 😢

loskaa, 2020-08-10:
-Tried all USB ports
-With dac and without it
-I have left those settings to default
-Tried with and without LCD
-I first hooked it using Roland XV sound module (in/thru)

I cannot try rest of your suggestions right now but I will when get another usb midi adapter (or I make a GPIO midi input). And I can probably try it with Raspbian and munt tomorrow.  

Thank you for using so much your time for this. I am beginning to believe that the problem might be in the uno. As it missing so much data. Who knows what have been changed during the years. I bought mine about a month ago.

dwhinham, 2020-08-10:
No problem! 😄 

It really doesn't sound like you're doing anything wrong, so I agree, I'm suspicious of the Uno.

Did you get it from a reputable seller and does it seem genuine? I haven't heard of fake Unos yet but there are a lot of faulty cheap/nameless interfaces out there so who knows, it wouldn't surprise me if they'd start putting them in cases that look like reputable devices! Either way, your experience with it might be a reason to add a caution to the "recommended list" in the README...

Let me know if you get any further. 👍

loskaa, 2020-08-10:
I bought it from gear4music and at least for me it seems genuine 😄 I’ll reply here when I know more.

loskaa, 2020-08-12:
Today I got a few octocouplers and rigged a GPIO midi input. And with it mt32-pi works perfectly. So I think it’s safe to say that the problem in my case was the Uno. 

Thanks again for the help!

dwhinham, 2020-08-12:
Ahhh great! Thanks for persevering, I'm glad you managed to get it working.
Maybe a different USB interface will work if you ever get a chance to borrow another one.

I'll close this issue now - enjoy! 😃

closed 2020-08-12 by dwhinham