dwhinham/mt32-pi #21: Hung notes

thorr2, 2020-08-16:
Hello, I just installed this for the first time and made my own MIDI adapter for the GPIO pins using your schematic.  I hooked up an OLED display and it works properly.  Munt is missing some note-off's and it is consistent (always happens in the same places in the same songs).  I am using Monkey Island 2 for example on my MiSTer and the music always has hanging notes in the same places that should not be there.  The display shows these notes hanging.  I have tried different chunk sizes (32 and 2048) and set my frequency to 32000 and the scaler to none.  Nothing so far seems to help.  Any ideas what could be causing this?  I have also played Doom on my MiSTer with my Roland SC-55 mk2 and it works fine with no hanging notes, so that is not the cause.  I also tried playing to the mt32-pi from my PC and it also has hanging notes.  Thanks for your help.

dwhinham, 2020-08-16:
Hi!

Which model of Raspberry Pi are you using?
Is there any chance you can get me a recording of the hanging notes or tell me a specific song/time to look for so I can try to recreate the problem?

thorr2, 2020-08-16:
Hello, I am using a Pi 4 model B.  Pretty much any song I play has the
problem.  I think the songs sound the same wrong way in the same places
every time I play them.  I have about 3 to 4 inch wires to the midi jack
and to the Pi GPIO pins from my midi breadboard.  I don't know if that
matters.

I tried installing Raspian with Munt to see if it was a hardware problem,
but I don't know how to get it to use the GPIO midi in port.

I will follow up with a link to a midi file and a recording of how it
sounds.

Thanks!
Michael

thorr2, 2020-08-16:
Hello,

Here is a midi file and a recording of it.  Notice the obvious stuck note
(there are probably others).  Right at the end of the recording the note
gets unstuck.  It always sounds the same.  When I play the midi file with
Windows sound instead of through the mt32-pi, that note is not stuck.

https://www.sendspace.com/file/yzvwc8

Thank you for your help!
Michael

dwhinham, 2020-08-17:
Thanks for sending me the MIDI.

I've just tested playing this song from Windows on `mt32-pi` v0.3.1, using my GPIO midi interface - it worked just fine with no hanging notes.

I'm guessing there is an electrical issue with the GPIO circuit.

- Can you measure the resistors to make sure they're the correct values and in-spec?
- Do you have another optocoupler you could try?
- Is your PSU and USB power cable good enough for the Pi? If you test with Raspberry Pi OS/Raspbian, you should not be getting any 'lightning bolt' symbols on the screen or undervoltage warnings in `dmesg` if your PSU is good and the Pi is under stress.

nswaldman, 2020-08-17:
I'm experiencing something similar with the PI-MIDI board I've just received: the onboard MIDI interface is doing drunk renditions of the MIDI data while USB is fine. Here's some listening material: https://www.dropbox.com/s/5kcjc7rlauv0scf/recordings-mt32pi.zip?dl=0

It's not identical, but there's a slim chance we're both having the same problem. I'm still in the process of ruling out the usual suspects (cables, power adapters, different Pi units).

thorr2, 2020-08-17:
Thank you for your help.  It is good to know I may be able to fix it.  I
have extra parts and can make a new midi board.  I will do more testing and
listen to and try the other person's example that is also having problems.

thorr2, 2020-08-17:
I listened to Nuey San's file and it seems to be missing notes.  I also
think I experienced this at one point.  It's possible there is a loose
connection or bad resistor or something.  I have a couple of questions:

- Are you both trying on a Raspberry Pi 4?
- Nuey San, when you play the file, does it sound identical every time you
play it, where it always misses the same notes?

Thanks!

thorr2, 2020-08-18:
Hello, I completely built a new MIDI interface with all new parts and I used a different set of resistors from a different manufacturer and measured each one before soldering it in to get the closest I could to the recommended values (and they were very close), and after everything was in place, I tried it and I still have the exact same problem.  I had this problem before and after installing a PCM5122 hat.  Are you testing on a Raspberry Pi 4 when you are having no issues?  Any further help would be appreciated.

thorr2, 2020-08-18:
I noticed this diagram uses different resistor values (220, 470 and 1K).  I am not sure if trying that would help.  https://www.instructables.com/id/PiMiDi-A-Raspberry-Pi-Midi-Box-or-How-I-Learned-to/

dwhinham, 2020-08-18:
I'd use that circuit diagram with caution, it has a pull up going to the +5V and the UART on the Pi is only capable of 3.3V, so there is risk of damage. Someone in the comments on that page also noticed this.

Sorry you're still having trouble, I'm really not sure what to suggest 😥

Can you provide hi-res photos of your circuit and how it's connected to the Pi?

The circuit in the README is the same as on this page: https://www.samplerbox.org/article/midiinwithrpi
I've tested it successfully on Pi 3, Pi 4, both constructed on a breadboard and on an assembled PCB.

Was your SD card 100% clean when you installed `mt32-pi`? (i.e. not just copied over an old Linux installation)

Can you try to test the interface in a clean install of Raspbian to try and rule out `mt32-pi`? You'll need to do the config dtoverlay changes and use ttymidi with something like MUNT or FluidSynth as suggested in this pagehttps://zuzebox.wordpress.com/tag/midi/

thorr2, 2020-08-18:
Thanks.  It is really late, so I will provide the pictures and work on this more tomorrow.  The SD card was clean formatted before installing the MT-32 Pi.  I noticed the Diode is different in the link you provided.  I am not sure if that makes a difference or not.  Also, I have a Pi 3 that I can try it on, but it will take some work to get to it (it's in a NesPi case acting as my Pi-hole).

dwhinham, 2020-08-18:
Uh oh - yes, you're absolutely right, it's meant to be a 1N4148.

Please try with one of those if you can!

thorr2, 2020-08-18:
I will have to order some.  Hopefully that will fix it!  Thanks!

dwhinham, 2020-08-18:
I hope so! 🤞🤞

If this was the reason, I apologise for the mistake - you're the first person to actually attempt to build the circuit from the schematic listed here and report back.

I'll fix the schematic ASAP.
Thanks!

nswaldman, 2020-08-18:
@thorr2 to answer your questions:

I've tested on a Pi3 and Pi4, the dropped/stuck notes are different for each Pi tested, but they seem consistent.

thorr2, 2020-08-18:
Sounds like we have the same problem.  I will hopefully have the new diode
tonight.  If it fixes it, maybe your pi hat also has the wrong diode.  I
will keep you posted.

thorr2, 2020-08-19:
I installed the new diode and it did not change anything unfortunately.  I guess I will try your suggestion of installing Raspian and trying to make that work.  I tried once before but didn't get very far.  Hopefully the link you provided will provide all the necessary information.  I also have a Pi 3 I can try, so I may try that first.

thorr2, 2020-08-19:
I tried the Pi 3 and it also had the issues, but less so.

thorr2, 2020-08-19:
This is the spec sheet for the 6N138 I used: https://media.digikey.com/pdf/Data%20Sheets/Lite-On%20PDFs/6N138-39%20Series.pdf
Is it possible this requires 5V and I need to buy a different one that works at 3.3V?

thorr2, 2020-08-19:
This is interesting (see the answer talking about 5V): https://electronics.stackexchange.com/questions/165255/midi-in-many-circuits

dwhinham, 2020-08-19:
Sorry the diode didn't change anything 😢 

From the datasheet, your opto should be fine with 3.3V, which to me means that that comment ("6N138 needs a 5 V power supply") does not apply here:
[image]

It supports -0.5 to 7V, and whatever you put in will be the same as what comes out, hence 3.3V into the opto means 3.3V out, safe for the Pi.

- You should still try Raspbian.
- You could try powering the opto via 5V as per the circuit in that StackOverflow answer, but you *must* add a voltage divider so that the signal is brought down to 3.3V before going into the Pi's Rx pin or you'll permanently damage the Pi.
- Try other manufacturers of opto aswell. I'm assuming you've ruled out the opto being faulty by replacing it with another unit?

thorr2, 2020-08-19:
Thanks.  I soldered in the 0.1 uF capacitor and changed the circuit to wire it up to the 5V on pin 8 with pin 6 going to 3.3V through the resistor per the diagram I linked.  It unfortunately did not change anything.  I used a different 6N138 when I made my second MIDI board but they were both from the same manufacturer.  I want to try Raspian, but it sounds like I will need to figure out all the steps from different sources.  I have a Pi 4, and that post was 2017.  Getting Munt to work with the GPIO MIDI IN is the challenge.  If I can't figure it out, maybe you can tell me all the steps in a list of what to do from scratch.  I can install Raspian from the imager tool and I am very familiar with Linux.
Thanks again,
Michael

dwhinham, 2020-08-19:
No problem, I'll do a quick write-up of how to test it in Raspbian shortly.

I've noticed another mistake - in my breadboard diagram, the connections from the MIDI DIN plug are reversed, and it's because the tool I used to draw it (Fritzing) has an incorrect pin mapping for the DIN plug pins - the schematic is OK though.

I doubt this will be affect you though, as you mentioned soldering and I guess you followed the schematic and not the breadboard. I'm still interested in seeing how you've built your interface!

thorr2, 2020-08-20:
Thank you very much.  I would really appreciate the write-up for what all to do in Raspian (including setting up Munt and using it with the GPIO midi port).  For the MIDI DIN, I went by the numbers on the schematic and the numbers embedded in the plastic on the MIDI connector.  I have uploaded pictures, so maybe you can check if it is right.  Here are the pictures:
https://www.sendspace.com/file/7kcr3i

Those are from my first midi interface that I made before changing it to 5V and changing the diode.  My second interface is soldered to the pins on the Pi Hat because I am trying to make it fit into a case and there is not enough room for the push-on connectors. They are basically the same though.

Thanks again!
Michael

dwhinham, 2020-08-25:
No worries, I'm glad you were able to follow them!

Unfortunately I don't know what else to suggest; my electronics knowledge only goes so far and I've just been using designs that others have shared. But at least we know for certain that it's a hardware issue in your case.

Hoping someone might chime in with a suggestion or feed back their own experiences, or maybe it's worth asking in that subreddit - I'd love to hear if you get any further!

thorr2, 2020-08-26:
Thanks!  My parts should be here by Friday.  I have two methods to try so
hopefully one or both will solve the problem.  I will keep you posted.

icb-, 2020-08-26:
> So, I guess the next step is to order a new 6n138 chip or try an alternative chip like the H11L1.

I would strongly suggest using the H11L1 (e.g. the [ON Semi H11L1M](https://www.onsemi.com/pub/Collateral/H11L3M-D.PDF)) over the 6N138. It has an open-collector schmitt trigger output, so still needs the pull-up resistor, but unlike the 6N138, the detector side is specified for a 3v to 15v supply voltage, so you can power it from the 3.3v supply without having to use a resistor divider to get the output to acceptable voltages. If you do change to the H11L1M, keep in mind that it's not pin compatible with the 6N138. Cost is a little higher, [91¢ for the On Semi H11L1M at Digi-Key](https://www.digikey.com/product-detail/en/on-semiconductor/H11L1M/H11L1-MQT-ND/284866) at qty. 1 compared to [81¢ for the Lite-On 6N138 at Digi-Key](https://www.digikey.com/product-detail/en/lite-on-inc/6N138/160-1795-ND/1969179), but negligible if you're only buying one.

The -0.5v to 7v for the 6N138 quoted above is misleading. That's from the "absolute maximum" section of the data sheet, that lists the voltages that can be applied without _damaging_ the chip, not the voltages it works at. A traditional 6N138 is specified to work with a 4.5v to 5.5v supply. Some chips may work from a 3.3v supply, but it's working outside its intended voltage range.

dwhinham, 2020-08-26:
Thanks for this info and clarification on my misleading interpretation of the datasheet. I couldn't find any better information to help @thorr2 on my own - sometimes, the only way to get the right answer on the Internet is to say the wrong answer so that someone will come along and correct you. 😄 

I guess 6N138s that work at 3.3v are just from blind luck. It'd be great to have a better example schematic up here based on the H11L1, so I'll order one too shortly and maybe we can improve the design.

thorr2, 2020-08-26:
Thank you Ian.  I have an H11L1M on the way so I will be able to try it.  I
did try running the 6N138 at 5v and pulling the output down to 3.3v and it
didn't help.  Hopefully the H11L1M will solve the problem.  Thanks again!

icb-, 2020-08-26:
Reading data sheets is a bit of a learned skill. They're often not as straightforward as they should be.

The [MIDI Electrical Specification](https://www.midi.org/specifications-old/item/midi-din-electrical-specification) allows for either 5v or 3.3v (added in 2014) from the transmitter. The recommended receiver circuit (ibid) includes a plain signal diode for reverse voltage protection, a current limiting resistor (as you would use with any LED), and a PC900V or 6N138 optoisolator (1). (The PC900V is a clone of the H11L1.) Since both optoisolators have open-collector (3) outputs, they also need a pull-up resistor.

Now that you know how the input is driven at the MIDI jack, you can look at the input specification for the optoisolator. The 6N138 [specifies](https://media.digikey.com/pdf/Data%20Sheets/Lite-On%20PDFs/6N138-39%20Series.pdf) 1.1v to 1.7v at 1.6mA (it also says 0.5mA on the first page, but doesn't specify the voltage requirements at that current) in the "Electrical Specifications" section. The H11L1 [specifies](https://www.onsemi.com/pub/Collateral/H11L3M-D.PDF) 0.75v to 1v at 0.3mA. So either one is perfectly acceptable on that front.

The MIDI specification says "The receiver must require less than 5mA to turn on." so both parts are acceptable there too. You need to select a current limiting resistor that meets that 5mA limit, and the 1.6mA for the 6N138, or the 0.3mA for the H11L1 minimums. There are also two current limiting resistor on the transmit side, each on the order of 220Ω for a 5v output, or 10Ω for a 3.3v output. Take those two resistors into account when calculating your current limiting resistor. In practice, you can probably pull more than the 5mA from a MIDI output, but don't push it too hard.

The MIDI specification also says "Rise and fall times should be less than 2 microseconds.". The 6N138 doesn't seem to specify its rise or fall time, but the MIDI specification says it has "been found to be acceptable." The H11L1 specifies 0.1µS.

Once you're past the "input" side of the optoisolator, you need to look at _your_ input's requirements. The 6N138 doesn't explicitly state its "output" voltage requirements in most data sheets, but you can get an idea of what it wants from the "test condition" column in the "electrical specification" section, that states the tests are at 4.5v. That makes it unacceptable for the 3.3v input on a Raspberry Pi. It **might** work from a 3.3v supply, but it might not. The H11L1 lists the "operating voltage range" of 3v to 15v in the "electrical characteristics" section.

MIDI runs at 31.25kbps. The 6N138 doesn't specify its switching time, but the MIDI specification says it's acceptable. The H11L1 specifies 1MHz, which is well above the needed baud rate.

Since the output of both optoisolators are open-collector, you need a pull-up resistor. MIDI is slow enough that you don't have to worry too much about the value of the resistor. If you use too high of a value, your rising edges will be too slow.

The shield and pin 2 (ground) on the MIDI connector should only be connected on the sender. Leave them unconnected on the receiver, or connected through small value capacitors to ground.

The [schematic posted above by @thorr2 from the TI forums](https://e2e.ti.com/support/microcontrollers/msp430/f/166/t/584083) is a good one to follow.

1. An optoisolator is essentially a LED and a photodiode or phototransistor in a single package. The 6N138 uses a darlington pair (two cascaded NPN BJTs) to amplify the signal. The H11L1 uses a Schmitt trigger (2) on its output stage. They're both open-collector. Being built of an LED and BJTs, the 6N138 is a current-mode analog device. That's not strictly what you want, but when driven with a logic signal, it gives a good enough logic output. The H11L1 has a proper logic output because of its Schmitt trigger.

2. A Schmitt trigger is a comparator with hysteresis. That is, it compares two voltages (in this case, the voltage present on a photodiode and an internal reference threshold) with hysteresis. Hysteresis means it has two threshold voltages. Between those two thresholds, the output will stay the same until the input crosses the other threshold. This stops spurious (unintentional) changes in output when the input is hovering around one of the thresholds.

3. An open-collector output can only pull to ground. You need a pull-up resistor to pull the output high. This is often used by multi-drop busses like I2C to prevent damage when two parts try to drive different logic states.

thorr2, 2020-08-26:
Thank you Ian!  In the schematic I posted, are all of the resistor values
shown ok for 3.3V?

icb-, 2020-08-27:
For the circuit on the TI forums post? Yes, the MSP430 it's drawn for is a 3.3v device too.

The resistor (R2) in parallel with the input reverse polarity diode (D1) can safely be removed. If you're trying to cut down parts, the reverse polarity protection diode can be removed with the worst that might happen is blowing out the diode in the optoisolator if you miswire the MIDI input (but not damaging anything past the optoisolator. That's what it's designed to protect.) The shield capacitor (C1) is completely optional, just reducing EMI. It will probably also work without the bypass capacitor (C2) on the optoisolator.

A minimal circuit is the XLR connector, the current limiting resistor (R1), the optoisolator, and the output pull-up resistor (R3).

Also note that the circuit shows two different grounds. Chassis ground (the 3 prong "fork" facing down) on the shield capacitor (C1), and signal ground (the downward pointing triangle) on the MCU and optoisolator. They would generally be tied together in your circuit

thorr2, 2020-08-27:
Great!  Thank you very much.  I will keep everyone posted how it goes.

thorr2, 2020-08-27:
I installed the new MIDI circuit (minus the capacitor on pin 2 of the midi connector going to ground).  When I installed the mt32-pi onto the micro SD card, the display shows pt32-pi v0.31 --> Init mt32emu (I2S) and stays there.  I am not sure what could be wrong at this point.  I will try installing Raspian again and see if I can get it to work.

thorr2, 2020-08-27:
I got it to work in Raspian.  I assume the mt32-pi code is trying to auto-detect the GPIO midi interface and doesn't like the new circuit for some reason, but it works in Raspian.  However, with the completely new circuit using the H11L1M chip, I STILL have the same problem with the same hanging notes.  The only thing I can think of is it doesn't like the 4 inch wires between the midi jack and my midi circuit and/or the 4 inch wires between my midi circuit and the GPIO pins.  There is nothing else to replace.  Or, there is some kind of software issue, but I tried my PI 3 previously and still had problems.  I guess I have to give up and buy a USB midi interface unless anyone has any other suggestions at this point.

thorr2, 2020-08-27:
I tried disconnecting R2 and changing R3 to 270 ohms.  It made no difference.

thorr2, 2020-08-27:
I have a Turtle Beach USB MIDI adapter and it worked fine in Raspian when I connected it to my Roland UM-One MK2 on my PC and MIDI played perfectly.  I then put the MT32-PI software back on there and I still can't get past the Init screen.  I am not sure why it won't get past the initialization anymore.  I tried connecting the Roland UM One to the Pi and it still won't get past the Init screen.

thorr2, 2020-08-27:
I disconnected the DAC and set it to PWM and it still hangs during the Init.  I can't even get it to work with USB at this point unless I use Raspian.  This is so frustrating. :-(

thorr2, 2020-08-28:
I forgot to put the MT32 rom files on there which is why it was getting stuck at the init screen.  Now I have the problem that the notes are cutting off too early when using USB.  I don't know if it is due to the hacked together temporary MIDI coupler I made or the Turtle Beach adapter.  I don't remember having this problem on Raspian.  I will order a proper coupler and go from there.

thorr2, 2020-08-28:
Swapping the Turtle Beach and the Roland UM one MK2 fixed it.  The Turtle Beach is plugged into the PC and the Roland is plugged into the Pi and it works without dropping the notes.  So I finally have a working solution.  I have no idea why I had problems using the GPIO MIDI input but at least this works.  Thanks for your help everyone.

dwhinham, 2020-08-28:
@icb- 
Thankyou for taking the time to write the extremely detailed replies with the MIDI electrical explanation - this is invaluable information. I still want to correct the sample schematic in the README, so I have received an H11L1 and will be testing it at some point over the weekend.

@thorr2 
> I forgot to put the MT32 rom files on there which is why it was getting stuck at the init screen.

You've just reminded me that I need to write an error message mechanism for the LCD!
As for MIDI, there's no detection for GPIO MIDI. The logic is: if `usb = on` in the config, scan for USB MIDI devices, pick the first one and try to enable it. If none found, assume GPIO. `mt32-pi` will not hang because of a missing MIDI interface, it'll just assume you have a GPIO setup and will wait for something to start talking on the RX pin.

So - final outcome is no GPIO circuit worked? What about when the Roland interface is sending to the GPIO circuit - did you try that? USB MIDI interfaces can definitely be flaky as we've seen in other issues.

thorr2, 2020-08-28:
Thanks.  The final outcome is every GPIO circuit sounded wrong in exactly
the same places leading me to believe something else is causing the issue.
It might be a timing issue or something like that.  I was using the Roland
interface the whole time until I started trying to use USB because I needed
two interfaces for that.  The Roland interface is the better of the two.  I
also tried it from two sources, my PC and my MiSTer, and both were fine via
USB and both were wrong but sounded the same exact wrong way every time I
played the same song via GPIO MIDI.

An error message on the screen would be great for the missing roms.  Will
it work with cm32l roms?  With what filenames? That was not talked about in
the documentation.

Thanks again!

dwhinham, 2020-08-28:
If you want to log what `mt32-pi` is receiving, follow this:
https://github.com/dwhinham/mt32-pi/issues/20#issuecomment-671571547

Run this build and play Pink Panther, then move the log file off the SD card, and then repeat a few more times.
Attach the log files here.

If we use a diff tool (eg. VS Code), it will prove whether the stuck notes are happening in the same place or not.

I'm sorry to say, but I doubt there is anything I can do here at all - I can't reproduce it, others are using GPIO just fine too, and the behaviour is similarly broken on an entirely different environment - everything points to some hardware issue/incompatibility with your particular setup or combination. There's hardly anything that can go wrong on the software side - `mt32-pi` simply feeds what's coming in on the serial port to the synthesizer.

`mt32-pi` should work fine with CM32L ROMs - anything Munt supports, we should support. Filenames stay the same though, the contents of the ROM and its hash is what changes the behaviour.

glaucon1984, 2020-08-28:
Hi,

I received my PI-MIDI from Aranet (not sure if I should put a link) and I set it up, I can confirm the issues from the first post. I can also hear issues with Monkey Island 1, the intro starts with a couple of sustained notes that end too early.

I've connected the MIDI output to a real Roland CM-500 (in CM-64 mode) and it works flawlessly, but MT32-PI seems to have issues.

I thought it might be my PI-MIDI board, but I'm wondering if there is any issue at software level. Can anybody else test the intro of Monkey Island 1 on MT32-PI and compare with one of the many videos in Youtube?

dwhinham, 2020-08-28:
This video on Arananet's channel is actually recorded by me: https://www.youtube.com/watch?v=w93uoiFhk-Y
I will record another video with the full Monkey Island 1 intro on my own Pi-MIDI shortly.

If you get the same faulty behaviour after following this troubleshooting guide (https://github.com/dwhinham/mt32-pi/wiki/Troubleshooting) then you should contact Arananet for a replacement.

dwhinham, 2020-08-28:
I constructed the circuit from https://e2e.ti.com/support/microcontrollers/msp430/f/166/t/584083 on breadboard, but **without** C2 and R2 as I didn't have them to hand. Please excuse the IO expander on the right, I'm trying to prototype control panel stuff and I only have one breadboard!

[image]

It works perfectly, just as well as the 6N138s I've been using up until now - no hanging notes. I could made another video but there's really nothing to see. Again, source is the USB Uno and ScummVM from the PC.

(@icb- I understand that C2 is probably for decoupling/noise suppression, but what purpose does R2 serve here? I haven't seen it in any of the other MIDI circuits - is it important?)

Braincell1973, 2020-08-28:
I have a data point midi board wired to the gpio, I'm getting major amounts of missed keypresses from 2 different midi keyboards, using 1 of the keyboards in USB midi mode worked perfectly its just the gpio thats wonky :(

glaucon1984, 2020-08-28:
Hi, I've done more research, and the PI-MIDI hat works fine using a real DOS machine. So the situation is quite surprising: MT32-PI+PI-MIDI works fine on a real DOS Machine, and MiSTer FPGA works fine on a real Roland device, but MiSTer FPGA with PI-MIDI+MT32-PI doesn't play well together.

MiSTer FPGA reaches a limit that real Roland hardware can tolerate but not this solution, and a real DOS machine with a proper MPU-401 speaks correctly enough to this solution that there are no issues, somehow **MT32-PI and MiSTer FPGA are not compatible between them.**

I guess this is such an edge case that it will be impossible to troubleshoot. I haven't tried your troubleshooting guide but I wonder if it's worth the effort.

dwhinham, 2020-08-28:
> I have a data point midi board wired to the gpio, I'm getting major amounts of missed keypresses from 2 different midi keyboards, using 1 of the keyboards in USB midi mode worked perfectly its just the gpio thats wonky :(

Hmm. Is this an `mt32-pi` issue though? Does it work properly under Linux?

> Hi, I've done more research, and the PI-MIDI hat works fine using a real DOS machine. So the situation is quite surprising: MT32-PI+PI-MIDI works fine on a real DOS Machine, and MiSTer FPGA works fine on a real Roland device, but MiSTer FPGA with PI-MIDI+MT32-PI doesn't play well together.

That's... odd, and interesting!

> MiSTer FPGA reaches a limit that real Roland hardware can tolerate but not this solution

What kind of limit? Sorry - can you explain?

> and a real DOS machine with a proper MPU-401 speaks correctly enough to this solution that there are no issues, somehow **MT32-PI and MiSTer FPGA are not compatible between them.**
>
> I guess this is [...]

I was sent this video of `mt32-pi` being used with a MiSTer - is their setup different to yours?
https://www.youtube.com/watch?v=o7yOY2bKXfk

Unfortunately I don't own a MiSTer and I don't fully understand all of the hoops they're jumping through to get MIDI out of the device - it's going to be very hard for me to help in this case.

Braincell1973, 2020-08-28:
I have a mister but I don't know the connection method so I can test that .

I think I'm going to slap a scope on the output from the data point although its puzzling that the usb works ok and several people have troubles with the gpio.......

The midi circuit is passive  being just a signal conversion so I think its unlikely its the datapoint board........

dwhinham, 2020-08-28:
> I think I'm going to slap a scope on the output from the data point although its puzzling that the usb works ok and several people have troubles with the gpio.......

An oscilloscope capture would be extremely useful, please feel free to share your findings. I'd be interested in seeing the waveform at the Rx pin.

I wish I had an oscilloscope so that I could see what my (working) setups look like.

Higgy69, 2020-08-28:
Hi,
I also just got my PI-MIDI. 
I was playing Monkey Island 2 (SCUMMVM on WinPC) and sounded good. I then tried AMIGA, sounded good (although not heard MIDI on AMIGA so no reference) but when you exit the game the last note played continues to play. Resetting AMIGA does nothing, you have to Power Cycle the Pi.
I was logging in to mention this when I read about Money Island 1, so I tried that and yes I am getting hanging notes!?

    Raspberry Pi 3 1.2GHZ
    UM-1EX Roland USB from PC running ScummVM
    mt32-pi v0.3.1 (latest release version)

Could it be the USB -> MIDI interface?
Also there are quite a few config options. I have tried 'Good' & 'Best' but no difference on Monkey Island 1 with hanging notes.
I also have a MiSTER but would really need a HD image for the A0486 core ready to boot with MIDI to test. @Braincell1973 - Once in the Core goto Core Menu and press right, there is URLINK or somthing like that, scroll through connection until MIDI is selected. My USB MIDI interface flashed and I could see the MiSTER connected on Pi-MIDI OLED.

Hopefully we will get there :) @dwhinham - your Money Island 1 sounds fine with PI-MIDI, so what are the variables:
USB MIDI
config file
Pi version (video is using a 4, do you have a RPi 3 1.2GHz? )
PSU? I am not sure. If it works on other RPi setups why would PI-MIDI be any different. Can't be a lot of extra current draw?

glaucon1984, 2020-08-28:
> What kind of limit? Sorry - can you explain?

I have no idea, maybe talks too fast for the Raspberry Pi, maybe it's too slow, I wish I had more knowledge about the topic. I have tried with and without SoftMPU, I have tried with and without "delayedsysex", it doesn't work well.

> I was sent this video of `mt32-pi` being used with a MiSTer - is their setup different to yours?
> https://www.youtube.com/watch?v=o7yOY2bKXfk

I'm not sure what USB-MIDI adapter are they using on MiSTer side, I'm using a Roland UM-One.
He's testing General Midi in Duke Nukem 3D, I've only tested native MT-32. I'd say the strange audio on Duke Nukem 3D soundtrack is caused by the mapping of MT-32 to General Midi.
I'm also using 31250 bauds.

The funny thing is that the MiSTer setup has an ARM CPU besides the FPGA and it has a mode that supports running Munt locally, but the notes sometimes don't come at the correct tempo, which we blame on the lack of CPU power, so I wanted to try an external solution that is small and easy to carry as the MiSTer.

Braincell1973, 2020-08-28:
I'm using a pi3a and datapoint in a custom 3d printed case with oled. The way I see it there are only a few variables affecting gpio...

Drive current affecting opto output.........
Opto speed......
Output signal condition.......
Mt32pi software ......

As there are clearly working gpio solutions its unlikely to be mt32pi, really leaning towards the midi interface output to gpio...........

dwhinham, 2020-08-28:
@Higgy69 I hope we'll get there too, I feel a bit useless here because I'm seeing all these failure reports but I don't know how I can help without being able to reproduce it... 😞 

> config file

Only things I changed were the minimum to get Pi-MIDI fully working: `output_device = i2s` under `[audio]` and `type = ssd1306_i2c` under `[lcd]`.

> Pi version (video is using a 4, do you have a RPi 3 1.2GHz? )

I only have a Pi 4 and a Pi 3 B+, the 1.4GHz version - they both work fine. However, I know that @nswaldman (who has problems) is using a standard Pi 3.

> PSU? I am not sure. If it works on other RPi setups why would PI-MIDI be any different. Can't be a lot of extra current draw?

I use a Sony Xperia phone charger (UCH32C), which does USB C power delivery; 5V 3A.
I've also successfully used a weaker PSU - probably too weak - with the Pi 3 B+ (an old HTC 1A phone charger).

@glaucon1984 
> The funny thing is that the MiSTer setup has an ARM CPU besides the FPGA and it has a mode that supports running Munt locally, but the notes sometimes don't come at the correct tempo, which we blame [...]

Ah yes, I was told about this. Munt is *very* CPU hungry, you definitely need over a GHz of ARM to reliably render all of the MT-32's polyphony properly, and I guess the MiSTer's auxiliary CPU just isn't quite enough.

**All:** I've started a wiki page here: https://github.com/dwhinham/mt32-pi/wiki/Compatibility-reports

It would be very helpful if you can add your results to it with as much detail as possible.

Thanks!

Braincell1973, 2020-08-28:
This is interesting I'm using rom....

Mt32_control.1988-09-30.vx.xx-patched.rom

Will try with the 1.07 rom tomorrow........

Higgy69, 2020-08-28:
@dwhinham - as you have a PI-MIDI a number of us should really be able to get to the same hardware setup as you.
Just tried with my RPi 3B+ 1.4Ghz and still have the hanging notes. Monkey Island 1 intro is really sounding bad compared to your video.  I am also now using an official RPi PSU. So I think we can rule out PSU.
Here is my editable config options in case something is different:

usb = on
output_device = i2s
sample_rate = 96000
chunk_size = 512
i2c_dac_address = 4c (00 also seems to work)
i2c_dac_init = none
resampler_quality = best
type = ssd1306_i2c
width = 128
height = 32
i2c_lcd_address = 3c

I am using the NON patched Control ROM and confirmed SHA1 with yours.
So what about USB - MIDI device is that now the main area of difference to focus on? Could there would be Windows variances? I am on Win10 64bit (have options of Win7 32bit)
My MIDI-USB is plugged into monitor's USB hub, pretty sure it is USB 2.0 as I use if for USB + MicroSD file transfer and get high MB/s.
My USB-MIDI (UM-1EX) is being used in ADVANCED DRIVER MODE - OFF This means it uses standard Windows Drivers. I can install Roland drivers but it is not needed after something like WinXP.
Does the MicroSD card do any buffering/access?  I am using an old (probably slow) 2GB formatted in FAT32.

Braincell1973, 2020-08-28:
For me I was using 2 different midi keyboards and playing notes by hand , both keyboards with midi over gpio can't get above 1 note per second ish before next note is missed,  using 1 of the keyboards in usb mode directly connected to the pi worked flawlessly........

dwhinham, 2020-08-28:
@Higgy69 

`i2c_dac_address` and `i2c_dac_init` have no effect on Pi-MIDI; it's using a DAC chip that needs no init sequence, so you can ignore these. completely.

What I can also tell you is that Pi-MIDI is using a circuit design that powers a 6N138 with only 3.3V (refer to the discussion with @icb-  re: electrical specs) way back up in the conversation. So that's a factor to consider (the chip is being operated out of spec).

As a side note, I have recommended that Arananet temporarily cease production of the Pi-MIDI until this issue is resolved, or at least that he build new units using H11L1s instead. Though please note that I am not in any way involved in the design, production or sales of this board - I can only try to provide software support.

Regarding Windows - standard Windows 10 here, USB Uno is using basic Microsoft built-in MIDI driver; nothing special.

SD card: The card is only accessed by `mt32-pi` at startup - to load the config and ROMs. After that there is no further disk access at all, so if it's attempting to play sound, it's successfully started up and the SD card can be taken out of the equation.

@Braincell1973

I recommend anyone with misbehaving MIDI interfaces to get a keyboard, or at least a **software keyboard** such as [Pocket MIDI](https://www.morson.jp/pocketmidi-webpage/). I've mentioned this in the [troubleshooting guide](https://github.com/dwhinham/mt32-pi/wiki/Troubleshooting):

> You could use a Windows PC with USB MIDI interface and Pocket MIDI to send MIDI notes by playing an on-screen keyboard. This way you can ensure that all key presses result in a corresponding Note On and Note Off pair. Try playing keys rapidly - if any Note Offs are missing, you probably have a hardware issue and missing/hanging notes will occur.

Braincell1973, 2020-08-29:
OK.... so changed rom to 1.07.

Hardware ..... 
Midisport 2x2 , win 7 scummvm 2.1.2
Pi3a and datapoint 6n138 board
128 x 64 oled
Pi pwm output

Working flawlessly in sq1, sq5 , dott.........

[image]
[image]

[35 more replies]

closed 2020-09-02 by dwhinham