dwhinham/mt32-pi #23: LCD oddness on certain games

raelik, 2020-08-30:
I'm not sure if this is actually an issue with the mt32-pi/munt implementation, or simply how the MT-32 works, but games that send custom instrument data up to the MT-32 (like Master of Magic) cause the LCD display to change and the numbers/volume disappear, leaving just the level meters.

raelik, 2020-08-30:
I know some games send cheeky little messages to the MT-32 display,  could it be something like this that's actually causing the issue?

dwhinham, 2020-08-31:
Yes, this is a feature - when `mt32-pi` (or a real MT-32) receives an LCD message SysEx command, it will show the message instead of the part status. There is a dedicated command for this, it doesn't have anything to do with custom instruments, but many games that are MT-32 aware will put up a message.

ScummVM also does this, you can see it happen at around 20 seconds into this video: https://youtu.be/iJcwZOR3k9Q?t=16

Can you clarify what's 'odd'? Is some game's message not working?

I don't know the real MT-32's exact LCD message behaviour, I'm assuming the message stays there forever or until an instrument is changed ... my LCD implementation is neither complete nor accurate, so any suggestions on how it should work would be appreciated!

Maybe I could add an option for messages to 'time out' after a few seconds so the display returns to the part activity.

raelik, 2020-08-31:
The only odd thing is that the display never goes back to the default.

In the videos I've seen of people using ScummVM with a real MT-32, the ScummVM message is only on the screen for a few seconds (https://youtu.be/TVfntgqRx1o), probably because it gets immediately followed by a message about which instrument it's using on one of the parts, though those are usually only shown briefly before it goes back to the normal part display.
You can see that happening with another game here https://youtu.be/PQt6CC5FFb4?t=330

raelik, 2020-08-31:
Oh, here's a video of the exact game I was trying on a real MT-32: https://youtu.be/fWHO77nMc5U

I think it may be these instrument messages causing the issue, as I don't believe that message ever displays on my mt32-pi.  It just blanks out the part number line and never shows it again.

dwhinham, 2020-08-31:
Right, instrument change messages are just not implemented yet. It's on the to-do list.

raelik, 2020-09-01:
No problem, just wanted to make sure I wasn't seeing something unexpected.

olliraa, 2020-09-03:
> Right, instrument change messages are just not implemented yet. It's on the to-do list.

I guess this applies also for displaying the current instrument name on the LCD in general?

dwhinham, 2020-09-03:
That's correct, yes. I'll add all of this in the next release as I can finally move onto more exciting features now that the MIDI note bugs have been fixed, and this will be quite a nice thing to have. 😃

dwhinham, 2020-09-19:
This has now been implemented in [version 0.5.0](/dwhinham/mt32-pi/releases/tag/v0.5.0). 😃

closed 2020-09-19 by dwhinham