dwhinham/mt32-pi #27: Feature Request: WiFi Samba Share

thorr2, 2020-09-13:
I have built a nice enclosure but it does not allow for ejecting the micro SD card without taking it apart.   It would be great if WiFi could be enabled to allow remotely editing the config file and changing the ROM files to the various versions of the MT-32.  Thanks for your consideration!

dwhinham, 2020-09-13:
I do want to implement something like this, whether something as complex as a SAMBA share can be done or not I'm not sure just yet, but a simple FTP server should be possible. Will look into it. 🙂

Every time I want to change even just a few lines of code I have to compile, copy, and swap SD cards - so believe me, I get it 😅

dwhinham, 2021-12-12:
FTP server implemented in v0.11.0.

closed 2021-12-12 by dwhinham

thorr2, 2021-12-13:
Awesome, thanks!