dwhinham/mt32-pi #30: No sound out- RPi4, PWM

FatalXCept10n, 2020-09-20:
I am using an RPi4 and have no audio output. I have a Fore USB MIDI interface. I extracted the latest release of MT32-Pi to the root of my SD Card, then copied in the two ROM files, named as the README specified. I could see the MIDI activity, MIDI OUT on the interface to MIDI IN on the keyboard, MIDI IN on the interface to MIDI out on the keyboard. I tried with the keyboard local synth enabled and disabled (Both of these should transmit over the MIDI interface according to the keyboard manual- it is a Yamaha). The speaker work when plugged in to other input sources, volume was properly set. I am wondering if there is a mixer or some way of setting the volume in MT32-Pi. I left the defaults for the audio entries in the file (All is set to PWM). File is attached but renamed as a txt so it will upload.
[mt32-pi.txt](https://github.com/dwhinham/mt32-pi/files/5252011/mt32-pi.txt)
I am not really a coder, but have been working with my kids on music, computers and Midi, messing with RPi, emulators, linux, etc. Let me know how I can help the project.

dwhinham, 2020-09-20:
Hi!

You won't hear anything unless you can make the keyboard transmit on Channel 2 or above.
This is a quirk of the MT-32 - on powerup it uses the channel range 2-10 - I should probably add this into the FAQ. 😃

There is an alternative mode that can be set via a SysEx command to make it use Channels 1-8 and 10 instead, but I don't have the magic incantation handy.

Hope this helps!

icb-, 2020-09-21:
> There is an alternative mode that can be set via a SysEx command to make it use Channels 1-8 and 10 instead, but I don't have the magic incantation handy.

AFAIK, there isn't a SysEx message to put the MT-32 into the channel remap mode. On a real MT-32, you do it by holding "master volume" and pressing "part 5", then pressing "part 1".

Without adding real buttons (I believe that's planned,) I think this needs either adding a mt32-pi SysEx message (and config setting?) that either remaps channels 2-9 to channels 1-8 in the MIDI parser, or simulates the required button pushes to Munt.

dwhinham, 2020-09-21:
You can use standard MT-32 SysEx - no need for a custom one; I worked it out last night. MT-32 SysEx just writes into various areas of RAM, and you can write new values directly into the channel mapping memory - this is also what the GUI version of Munt does.

I'll eventually add that button combination when we get there! 😄 

This SysEx string will configure the channels to be mapped to 1-8 and 10:
```
F0 41 10 16 12 10 00 0D 00 01 02 03 04 05 06 07 09 3E F7
```

...and this will restore the channels back to power-on defaults:
```
F0 41 10 16 12 10 00 0D 01 02 03 04 05 06 07 08 09 36 F7
```

The first 5 bytes are some preamble, `0x1000D` is the address of the first MIDI channel map memory location as shown in the MT-32 manual:
[image]

The next 9 bytes are the MIDI channels to use, and finally there's the usual Roland checksum byte and the End of SysEx byte (`F7`).

I've created and attached two .SYX files that you can send with many popular MIDI tools:
[mt32_channel_remap_sysex.zip](https://github.com/dwhinham/mt32-pi/files/5255269/mt32_channel_remap_sysex.zip)

@FatalXCept10n 
If you can't get your keyboard to output on Channel 2 or above, use a computer to send `mt32_alternate_channels.syx` from the archive above to your Pi using MIDI-OX or similar on a PC, and then you should be able to play sounds on Channel 1.

dwhinham, 2020-09-30:
Closing due to inactivity - feel free to reopen if there are any problems.

The SysEx/Channel remap info [has been added to the wiki](https://github.com/dwhinham/mt32-pi/wiki/MIDI-channel-assignment) as well as the FAQ.

closed 2020-09-30 by dwhinham