dwhinham/mt32-pi #31: Change default channel inputs

T0lk13n, 2020-09-21:
Hello.
First of all thanks for this great piece of code :)

And now a question.
MT-32 use by default inputs 2-9(and 10 to percusion) but it can be changed to 1-8(10 remains the same) with a keys combination.

Is there a way to can do it with this device? Perhaps sending a sysex msg?

Thats all and thanks again.

dwhinham, 2020-09-21:
> First of all thanks for this great piece of code :)

You're most welcome! It's standing on the shoulders of the excellent Munt core though, so we have the team behind that project to thank too! 😄 

> MT-32 use by default inputs 2-9(and 10 to percusion) but it can be changed to 1-8(10 remains the same) with a keys combination.
> 
> Is there a way to can do it with this device? Perhaps sending a sysex msg?

Yes, there is - coincidentally this was discussed in another issue. Have a look at [this reply from me](https://github.com/dwhinham/mt32-pi/issues/30#issuecomment-696082355) to find some SysEx files you can send to the Pi to remap the channels. You can use a Windows tool such as MIDI-OX to send it, or any other generic MIDI tool capable of sending SysEx.

Would it be useful to have a config file option to set the default channel mapping on startup?

T0lk13n, 2020-09-21:
Oh man! I promise i have looking for this kind of info all this weekend.
Thanks so much.
But yes, I think an option to change this would be great.
Ah! And update the faq to let know this issue with default cahnnels. I thought my friend Edu Arana did a faulty device ;) sorry Edu, you are the best!

dwhinham, 2020-09-21:
No problem!

I've now updated the FAQ, and created a new wiki page with full details (including the info I linked): https://github.com/dwhinham/mt32-pi/wiki/MIDI-channel-assignment

Hopefully this will help avoid some confusion for future users who are new to MT-32. 😃 

I'll keep this issue open as a feature request for the config file idea.
Thanks!

closed 2020-09-22 by dwhinham

dwhinham, 2020-09-22:
I've added the config option feature; it'll be available in the next release, or [here](https://github.com/dwhinham/mt32-pi/actions/runs/267543247) if you want to try it now. 👍

commit 2b7c0e9c (2020-09-22): Add ability to configure MIDI channel assignment

Closes #31.