dwhinham/mt32-pi #33: General MIDI mt32 official compatibility 'patch'

jsmolina, 2020-09-28:
Hi,

I was wondering how these affect general midi compatibility.
http://www.houseofsynth.com/roland-mt-32-to-general-midi-conversion-utility
or here (more official)
http://cms.rolandus.com/assets/media/exe/MT2GM.EXE

It might need to simulate playing these midi files upon device boot? I would be GREAT to have Duke Nukem or Doom playing on its whole glory.

dwhinham, 2020-09-28:
From what I can tell, they're ***okay*** for some purposes, but no replacement for a true GM synth.
Have you actually tried them?

Personally, I'm reluctant to integrate these files into the project for the following reasons:

- It's arguably easy enough for the user to just play the MIDI file from their host machine already. You can just play it with e.g. [DOSMID](http://dosmid.sourceforge.net/) before starting Doom.
- I'm unsure of the license/copyright on these files. We can't just cherry-pick random files into the project without knowing the license. 😅
- We can offer a far superior General MIDI experience with FluidSynth, which is planned (see #19).

If there's an overwhelming demand for this feature, then I could maybe think of something, but it'd be a very low priority I think.

Thanks!

raelik, 2020-10-10:
I've used MT2GM to play Doom, sounds spot on, but that's only because Doom doesn't use that many channels.  If you use it to try to play something with music intended for an SC55 or 88, and uses more channels than available on an MT-32, or instruments that it doesn't support, then things will get weird.

jsmolina, 2020-10-13:
@raelik could you share which files did you use before switching to General MIDI mode? I tried but it didn't sound really ok on doom.

https://www.youtube.com/watch?v=vA2G5wPOSdo&t=7s

raelik, 2020-10-13:
@jsmolina I grabbed MT2GM ("MT-32 to GM") here: https://www.midimusicadventures.com/queststudios/mt32-resource/utilities/

I'm also using GSPLAY instead of PLAYMID, but either should work.  The files I'm using from the self-extracted zip are MTGM.MID and MTR-STND.MID, and I simply play them through the mt32pi using GSPLAY before playing Doom.  After playing, you either need to hard reset the mt32pi (or send a SysEx reset) to return it to normal MT-32 mode.

dwhinham, 2020-10-13:
Thanks for helping each other with this!

This information would be great material for the wiki if anyone fancied helping write a page about General MIDI and how to use this patch.

I think there's quite a few users new to MT-32 who don't understand how this patch works or about GM/MT-32 (in)compatibility who could benefit a lot from this 🙂

jsmolina, 2020-10-15:
> @jsmolina I grabbed MT2GM ("MT-32 to GM") here: https://www.midimusicadventures.com/queststudios/mt32-resource/utilities/
> 
> I'm also using GSPLAY (also available on that page) instead of PLAYMID, but either should work (I think so anyway, GSPLAY definitely works). The files I'm using from the self-extracted zip are MTGM.MID and MTR-STND.MID (standard rhythm instruments), in that order, and I simply play them through the mt32pi using GSPLAY before playing Doom. After exiting the game, you either need to hard reset the mt32pi (or send a SysEx reset) to return it to normal MT-32 mode.
> 
> And yeah, the music in your YouTube video isn't with the GM patch properly applied. That's how it sounds without it. I've uploaded a sample that I recorded through an audio interface from my mt32pi with the patch properly applied: https://gist.github.com/raelik/86319680300f716ee35148d250a22f1e/raw/cf6e1f9887ac08b255951fb9584cbc5f3672a1a8/doom_mt32-pi.mp3 (skip to 1:28 if you just wanna hear E1M1)
> 
> Dunno how I missed that your video was Doom 2, not Doom 1. Regardless, I went and double-checked, and definitely, your video is how it sounds without the patch properly applied (I don't have a recording of it, but I checked with the patch applied and without).

Now sounds better! 
https://www.youtube.com/watch?v=8pZbP9AQGS0

dwhinham, 2020-10-15:
Ahh yes; "Sound Canvas" will be using GS, which is Roland's extension that goes beyond General MIDI, so that would explain it.

Sounding better! :+1:

dwhinham, 2020-11-22:
I just released [v0.7.0](https://github.com/dwhinham/mt32-pi/releases/tag/v0.7.0) with FluidSynth support, so I think we can close this. 🙂

closed 2020-11-22 by dwhinham