dwhinham/mt32-pi #36: USB MIDI plug & play (hotplug) support

dwhinham, 2020-10-04:
As of Circle Step 43, USB plug & play is supported, but requires the application to make use of it.

This could be useful for temporarily connecting other MIDI equipment, or moving an in-use USB interface to a PC without having to restart `mt32-pi` after disconnecting/reconnecting the interface.

sigkill, 2020-10-16:
I just tried with a very cheap USB midi interface, and a very nice RK-004 with MIDI port 1 set to IN, was not able to get further info as the terminal is disabled / and no display.

Am working with a Pi 3 Model B with the last release.

I've also got a few higher end boards:
HiFi Berry - PCM5122
AudioInjector.net - V. 1.5 AIH with WM8731S
PiSound from blokas with midi in and out, and a PCM1804 chip, and a few other ICs I don't have the ability to read.

Is there anything I can do to get to the console, or terminal to help out with the project? I would love to help out, and have at least one of these working, if not all three.

dwhinham, 2020-10-16:
Hi!

I appreciate you wanting to test all this hardware (you have some nice stuff there that I don't think anyone else has tried) - but this isn't the right issue ticket to discuss it. 😅

This issue is about being able to connect/disconnect/reconnect a USB MIDI device while the power is on and `mt32-pi` is running. Right now, USB devices have to be connected before powering up, and you can't disconnect/reconnect them without power cycling.

Would you mind opening a new issue and we can chat about getting these working? You can just literally copy and paste your previous message. I can provide you with a console-enabled build.

Thanks!

sigkill, 2020-10-17:
Sure can do - thank you!

commit 7f368cd8 (2021-01-31): Enable USB plug and play support

Closes #36.
commit 2280a5a5 (2021-02-03): Enable USB plug and play support

Closes #36.
closed 2021-02-07 by dwhinham