dwhinham/mt32-pi #37: USB MIDI interfaces - Roland UM-1EX - Working

Higgy69, 2020-10-10:
Hi,
Just to say this is the device I have been using, so it can be added to the 'Working List':

https://www.roland.com/global/products/um-1ex/

Thanks
P.S. Just updating to v0.6
P.P.S. If/when control buttons are added, I plan to hack them onto my PI-MIDI. The GPIO pins are accessible on the top of the PCB for soldering thin wires. I suppose a 'rotary controller' is the easiest as you get 3 inputs allowing al sorts of menu control with minimal wires. Works well on 'FlashFloppy' https://github.com/keirf/FlashFloppy/wiki/Hardware-Mods#rotary-encoder

dwhinham, 2020-10-10:
Thanks!

To clarify, are you attaching this device to the Pi's USB ports, or using it to send data from a PC to the PI-MIDI.
The intention of the compatibility table is for using a USB MIDI interface to *receive* data on the Pi (I should clarify that really).

Re: encoders, yes; this is already being discussed in #16.

Cheers 😃

Higgy69, 2020-10-10:
Hi.
I am using it to send data from my PC and MiSTer FPGA to PI-MIDI.
I will check out #16.
Cheers.

dwhinham, 2020-10-10:
I've added this info to the wiki (I'm going to start moving some stuff from the README to the wiki soon instead).
https://github.com/dwhinham/mt32-pi/wiki/USB-MIDI-interfaces

Thanks again 🙂

closed 2020-10-10 by dwhinham