dwhinham/mt32-pi #38: Show selected instrument instead of channels status?

olliraa, 2020-10-11:
This software is so cool! Thank you for your great efforts on this, really appreciate! For live playing it would be cool to have the option to have the display show the selected instrument instead of the midi channel status. 

Would this be possible? :)

This is soo great :D

dwhinham, 2020-10-12:
Thankyou, glad you are enjoying it! 😄

I **think** this will come naturally as part of implementing #16, as on a real MT-32 you can hit a button to enter patch select mode and you can see and change the instrument assigned to the channel whose button you pressed.

I have a real MT-32 now, so this will only be a matter of time.

Otherwise as a temporary solution, I could do a quick hack build for you that always shows the instrument if that helps!

olliraa, 2020-10-13:
If you already haven't started creating a temporary build, there's no real hurry. I'll be patient :) Thank you for great communication!

icb-, 2020-10-13:
Is there enough space to keep the selected patch names on screen? `mt32emu_qt` shows them, but of course there's a lot more space to work there.

Text modules are way too small. You only have room for 2 or 4 names on the 20x2 and 20x4 displays.

The supported OLED displays don't fare much better. Using a tiny 8x8 font, you're only going to get 4 rows of 16 characters on a 128x32 OLED display, so you can only see half of the non-rhythm channels. It looks like the longest patch name is 16 characters by itself, so you can't fit any legend or status information.

If you used the 3x5 [Tom Thumb](https://robey.lag.net/2010/01/23/tiny-monospace-font.html) you could fit 5 rows of 32 characters, but that's barely readable on the 0.91" modules. If you use two columns, you could fit all of the patch names with no legend or status information.

With Tom Thumb and a 128x64 OLED display, you could fit all of the patch names on 10 rows of 32 characters, leaving a little room for legend and other status information, but again it's just barely readable.

dwhinham, 2020-10-13:
Yeah; I'm not so sure I want to go to those lengths for such a niche feature - as you say the font size would make it barely legible. Bigger graphical displays could be a (very) long-term thing, but it wouldn't be a priority,

@olliraa 
I've just done a maintenance release (v0.6.1), and while I was at it, I added this hack branch to keep the instrument on the screen. Download the sdcard package from here:
https://github.com/dwhinham/mt32-pi/actions/runs/305121903

I don't really want to implement this as a config option or a special feature, but this may tide you over for live playing until full MT-32 control surface simulation comes about.

Hope this helps 🙂

olliraa, 2020-10-14:
Works like a charm :) Thank you!!!

mrglennjones, 2023-07-16:
> Yeah; I'm not so sure I want to go to those lengths for such a niche feature - as you say the font size would make it barely legible. Bigger graphical displays could be a (very) long-term thing, but [...]

seems this link is dead now, how can i get the instrument names displayed please?