dwhinham/mt32-pi #39: Testing some additional DAC Hats PiSound, HiFi Berry, AudioInjector.net

sigkill, 2020-10-17:
As requested - opening a new issue:

I would like to test the following hardware.
HiFi Berry - PCM5122
AudioInjector.net - V. 1.5 AIH with WM8731S
PiSound from blokas with midi in and out, and a PCM1804 chip, and a few other ICs I don't have the ability to read.

Is there anything I can do to get to the console, or terminal to help out with the project? I would love to help out, and have at least one of these working, if not all three.

dwhinham, 2020-10-17:
Hi 🙂

Have you tried any of the hardware yet? Which Pi model are you using?

I'd start with MIDI, because you can get sound output (albeit poor quality) from the Pi's headphone jack. Once you know you can trigger the synth (hint: Pi's status LED flashes when MIDI is being received) and get sound output, you can move onto DACs. You can just use a USB MIDI cable to start with, but if you only have one of the cheap/nameless ones, you almost certainly will have problems - [see the wiki page](https://github.com/dwhinham/mt32-pi/wiki/USB-MIDI-interfaces).

### PiSound
The PiSound's site says:
> MIDI on Pisound is implemented using high speed SPI and a dedicated MCU for translating SPI data to serial MIDI byte streams and it's readily recognized in audio software as an ALSA Raw MIDI device.

This is the first device I've heard of that uses SPI for MIDI, and I don't have any SPI support yet in `mt32-pi`, though it should be possible. It'd be nice to get it working, but I don't have one, so I don't think I can help with this unless someone were to send/loan me one.

The PCM1804 on the PiSound is an ADC (input) for recording. Recording is beyond the scope of this project, so it won't be supported.

The PiSound's DAC (output) should work though, as it seems to be just using a PCM5102A which we know works, so it'll just be a case of enabling I2S in the config.

### HiFiBerry / AudioInjector.net
The HiFiBerry should be already working, although there's a small chance you'll need to find its I2C address, because this chip requires initialization. Please carefully read the [DAC section of the README](https://github.com/dwhinham/mt32-pi#-i2s-dac-support) to see how to enable DAC support. You could start by trying the settings in the table for **Pi-Fi DAC+ v2.0**; it might just work; otherwise see the section on how to find and set the correct I2C address.

Same goes for the AudioInjector.net card - I see that the WM8731S connects to I2C too, so it may require initialization,. Again, try enabling I2S audio - if it's silent, then we may need to do some tricky stuff to wake it up/unmute it.

### HDMI console
Here are some kernels with HDMI enabled. Since you didn't say which Pi you are testing on, I've compiled for both Pi 3 and 4.
Note that with these kernels, the Pi 4 will freeze if you don't connect a screen. That's why HDMI isn't normally compiled-in (this is in the FAQ).

[kernels-6.0.1-hdmi.zip](https://github.com/dwhinham/mt32-pi/files/5396299/kernels-6.0.1-hdmi.zip)

Note that I'm not sure what you were hoping to see - the only auto-detection/logging that happens is for USB, and you may get an error if an I2C device didn't respond when enabled in the config, but that's about it. It's not like Linux `dmesg` where you see devices being probed etc.

Hope this helps!

sigkill, 2020-10-19:
Sorry about that - A Pi 3 and a Pi 4 (I have both, will test both.)

dwhinham, 2020-10-19:
No worries! Also note that in v6.0.0 and v6.0.1, USB MIDI receive support (i.e. plug USB MIDI interface into the Pi) is broken - I just released v6.0.2 to fix that, but I need to update the HDMI kernels for that if you need them.

sigkill, 2020-10-20:
The midi usb I have shows as this on another pi running raspbian:
Bus 001 Device 003: ID 1a86:752d QinHeng Electronics CH345 MIDI adapter

modules loaded are:
lsmod | grep midi
snd_usbmidi_lib        36864  1 snd_usb_audio
snd_rawmidi            32768  1 snd_usbmidi_lib
snd_seq_device         20480  1 snd_rawmidi
snd                    90112  14 snd_seq_device,snd_hwdep,snd_usb_audio,snd_usbmidi_lib,snd_bcm2835,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi

Linux Hardware Info listed here too:
https://linux-hardware.org/index.php?id=usb:1a86-752d

dwhinham, 2020-10-20:
Thanks, but I'm not sure what you want me to do with this information?
Have you actually tried any of your hardware with mt32-pi? Does it work? 

More info please.

I know what this USB device is, it's a known-bad USB MIDI device, they are broken by design. See the wiki:
https://github.com/dwhinham/mt32-pi/wiki/USB-MIDI-interfaces#cheapgeneric-usb-interfaces

If you want to try and modify the USB driver to make this cheap device work better, you need to contribute to https://github.com/rsta2/circle, not this project.

sigkill, 2020-10-20:
Thanks for the link to the wiki, and other projects. I've purchased a UM-ONE mk2 as that is listed as working, should be here in a few days. I wanted to list out what my test hardware specs were as with the two non PiSound hats - I would need to use an external MIDI source via usb.

Will update you in a few days when I have a good testing platform for both Pi 3 and Pi 4 ... For what it's worth I've attempted using the different I2C addresses on the Pi 3 with the AudioInjector and the HiFi Berry, but I suspect it was the garbage cheap midi usb that was preventing testing.

I am in a holding pattern until I can spend some time Friday or Saturday checking out all these things in a known to be okay working hardware configuration using the Roland UM-ONE mk2.

dwhinham, 2020-10-20:
Ahh okay, I understand 🙂

The UM-ONE Mk2 is probably the gold standard of interfaces; it's not cheap, but it's very reliable - good choice.

No rush with the other stuff! 🙂

sigkill, 2020-10-22:
UM-ONE Mk2 came today - I think there may be a hotplug / hdmi safe option for the config.txt that can be used in conjunction with an overlay for the Pi 4, so I will test that option as well. Probably will test after work today as soon as I get some free time. Wanted to give an update as I am excited to try it out.

sigkill, 2020-10-22:
Okay - still appear to be having some kind of USB issue. I've connected and validated that the UM-ONE Mk2 works in Raspbian, and works on other devices, as well as the Pi 3 and Pi 4 when booted under a normal OS.

When I boot using the HDMI kernel provided I see the following on the display:
[https://imgur.com/a/FkioPer](url)

The errors present lead me to believe the USB isn't getting initialized correctly, as there are a few errors reported for usbdev-0-1 and the dwhci errors that follow.

Under raspbian, on the same Pi 3 I was able to connect to the Roland UM-ONE Mk2, but the errors after booting appear to have some kind of issue with the usb ports. I tried shutting off and turning on in each port, without luck.

May I ask if there are any flags that could be set, or if the kernel config perhaps has some option that skips over the initialization of the usb ports too quickly?

dwhinham, 2020-10-22:
I can see youre running 6.0.1 - you missed this from above:

> Also note that in v6.0.0 and v6.0.1, USB MIDI receive support (i.e. plug USB MIDI interface into the Pi) is broken (issue #40) - I just released v6.0.2 to fix that, but I need to update the HDMI kernels for that if you need them.

sigkill, 2020-10-22:
Thank you!

I see a red light now and no errors. Will proceed with tests as soon as I am off work today.

dwhinham, 2020-10-22:
Great! You should at least be able to get one of your DACs working - let me know how it goes! 🙂

sigkill, 2020-10-23:
HiFi Berry DAC+ confirmed working (Version 2.0) Address was 4d, confirmed from a post on their forum. As you stated earlier, these appear to match the two other similar devices from the wiki.
mt32-pi.cfg: deltas:
output_device = i2s
i2c_dac_address = 4d
i2c_dac_init = pcm51xx

AudioInjector.net Device: So far no dice. I haven't been able to get it to work, am going to setup a generic raspbian install on another sdcard and check addresses etc.

https://github.com/Audio-Injector/stereo-and-zero
The kernel overlay was worth a look, apparently there is a similar device or chip mentioned here: https://github.com/lineage-rpi/android_kernel_brcm_rpi/blob/lineage-17.1/arch/arm/boot/dts/overlays/audioinjector-wm8731-audio-overlay.dts 

Have not progressed on the Pi Sound just yet, but the stuff that works sounds great!

I'm using a Korg Micro XL as an input device and tested channels 2-10. Might setup a STE or Falcon and give a test on Monkey Island later this weekend.

sigkill, 2020-10-23:
Forgot to add that I will run same testing on the Pi 4 later tonight too.

dwhinham, 2020-10-23:
Excellent! Glad the HiFiBerry works well, I'll get that added to the README compatibility table tonight, thanks for confirming!

Ahh, now I see what you mean by the overlays - yes, the sources for these can offer useful clues sometimes. 🙂 My apologies, I thought you were about to start trying to use the .dtb's with mt32-pi!

We should be able to get AudioInjector going - if we need a custom init sequence, I'll write up some stuff for you to try.

The way I determined the init sequence for PCM51xx was to dump all of its I2C registers without the Linux driver loaded, then dump them all again with the driver loaded, and compare. Then by checking the datasheet we can verify what the register changes mean.

You can do this with `i2cget` and a script to loop over all the interesting registers - if that's enough for you to go on, go for it! Otherwise I can help you with the steps.

Thanks again!

mattclaw, 2020-11-14:
Wanted to chime in to mention that I can also confirm working status for [HiFiBerry DAC+ Pro](https://www.hifiberry.com/shop/boards/hifiberry-dac-pro/).

dwhinham, 2020-11-14:
Thanks for the confirmation! 🙂

dwhinham, 2021-02-14:
Closing this as we now have Pisound support at least - if we want to revisit testing unnconfirmed DACs we can re-open this issue or make new ones.

closed 2021-02-14 by dwhinham