dwhinham/mt32-pi #40: USB Midi Input (pi) Broken in 0.6.x

calvinmorrow, 2020-10-18:
USB Midi Receiving seems to be broken on the pi side in both 0.6.0 and 0.6.1.

Raspberry pi4 with USB plugged into Roland UM-ONE (also tried with M-Audio Uno) show Midi IN flashing with no sound and no messages on an attached 1306 OLED.

Audio is being sent via SoftMPU testing via Space Quest 3.

Release 0.5.0 is working as intended.  Status Display messages show up (Insert Buckazoid, Space Quest III, and Channel meters all working).

Same config file was used testing all releases.

dwhinham, 2020-10-18:
Wow, you are absolutely right. How on earth did I miss this?!

I'm so sorry 😔

I'll get this fixed ASAP - thankyou for the report!

dwhinham, 2020-10-18:
I've found the problem (and a solution). There's a small issue in Circle and I've proposed a fix - in the meantime I'll prepare a hotfix for `mt32-pi` to work around this.

Thanks again.

commit d05bd1f2 (2020-10-18): Temporarily disable GC_SECTIONS

Using GC_SECTIONS results in broken USB device numbering since Circle
Step 43.

Disable until rsta2/circle#143 is merged or an alternative fix is
implemented.

Fixes #40.
closed 2020-10-18 by dwhinham