dwhinham/mt32-pi #42: Pi Sound Support

sigkill, 2020-10-23:
Let me know where to send a PiSound, after reading some of the other things you're working on, I think it would benefit both projects to send you one for development purposes.

dwhinham, 2020-10-23:
Awesome! Thankyou for this - I've replied to your DM! 🙂

dwhinham, 2020-11-28:
I received a Pisound from @sigkill last week and I've got it working! 😀
Many thanks for getting one to me!

A test build is available here: https://github.com/dwhinham/mt32-pi/actions/runs/389044086

It should be automatically detected - all you need to do to the config file is ensure that:

- `output_device = i2s`
- `sample_rate` is set to either `48000`, `96000`, or `192000` (the only rates supported by the Pisound

You'll know it's been detected because the MIDI IN/OUT LEDs will flash once on startup.

sigkill, 2020-11-29:
Thank you! Tested it with the settings and kernel from the https://github.com/dwhinham/mt32-pi/actions/runs/389044086 test build (the one that was included in the build worked.) I tested on a Raspberry Pi 3B+ with a version 1.0 PiSound card, and an Atari 1040 STE. Both were able to send MIDI. Rates 48000, 96000, 192000 all confirmed.

Awesome job! Thank you for getting this working.

dwhinham, 2020-11-29:
Excellent, thanks for testing!

Meanwhile, the patch to Circle required to get this working has been upstreamed: https://github.com/rsta2/circle/pull/167

I'll include this driver in the next release. 😃

commit e76f1eee (2020-12-02): Initial support for the Blokas Pisound

This commit adds a driver for the Blokas Pisound audio/MIDI HAT based on
the Linux sources.

Currently this requires a patch to Circle to allow the Pi's I2S audio
device to operate in slave mode.

The Pisound only works at 48000Hz, 96000Hz, and 192000Hz. The Pisound is
automatically detected by the startup code; the user only has to enable
I2S audio output.

MIDI output and recording functionality is not supported.

Closes #42.
commit 4de13be8 (2020-12-08): Initial support for the Blokas Pisound

This commit adds a driver for the Blokas Pisound audio/MIDI HAT based on
the Linux sources.

Currently this requires a patch to Circle to allow the Pi's I2S audio
device to operate in slave mode.

The Pisound only works at 48000Hz, 96000Hz, and 192000Hz. The Pisound is
automatically detected by the startup code; the user only has to enable
I2S audio output.

MIDI output and recording functionality is not supported.

Closes #42.
closed 2021-01-03 by dwhinham