dwhinham/mt32-pi #43: 'Break Error' on display when powering on PC

Higgy69, 2020-11-15:
Hi,

MT32 Pi has been working great on a number of systems. On my old PC I have a Yamaha YM724 card which runs great with SoftMPU.
I just got another soundcard 'Fortemedia FM801' this has been listed as working with SoftMPU. I have the software detecting everything, but as soon as I power on the PC I get 'Break Error' appearing on the MT32 Pi display.
I swapped back to the Yamaha FM724 card using same Gameport MIDI lead, Micro SD config etc and it all works again?
It seems irrelevant of PC software settings as it appears as soon as PC is powered on.

Thanks

dwhinham, 2020-11-15:
Did you try playing any music?

You can probably ignore the Break Error, it's just a warning that the serial port was in an unexpected state, which is probably normal when the machine is starting up and the gameport is being initialized.

If it works, just ignore it.

Higgy69, 2020-11-15:
I tried Monkey Island 2. The display does not show any Lucas Arts message and there is no sound.

 I am hoping the Gameport is not wired differently, it does say in the manual MPU401 UART mode MIDI interface joystick. So hoping it is same as other Gameports.

If the warning message is nothing to worry about I can concentrate on hardware/motherboard settings. I have another old PC I can try it with.

Higgy69, 2020-11-16:
I have fiddled around with BIOS settings etc and tried another PC but no luck.
I did find a very comprehensive Datasheet on the chip, so I don't know if there is something in there which might help?
https://datasheetspdf.com/pdf-file/853822/ForteMedia/FM801-AU/1

I will also try to reach out to the person that said SoftMPU works on this card (the software detects everything).

Thanks

dwhinham, 2020-11-16:
This doesn't sound like an mt32-pi issue - try to establish whether you can get this card to send valid MIDI (with any other MIDI synth perhaps, it doesn't have to be an MT-32).

"Break error" is caused by the Raspberry Pi's hardware UART controller saying: "The thing connected to me is sending garbage" - as in, not even decipherable data, let alone MIDI data, so I think you need to focus on the PC and SoftMPU.

Try installing the card in a Windows machine and send MIDI with a Windows app; that'd rule out the card.

icb-, 2020-11-17:
Do you have any other MIDI hardware? Testing whether you can receive MIDI messages from the card on something else would be a good step to determine where the problem is.

Do you have any electrical test equipment? An oscilloscope or logic analyzer would be ideal, but even a simple DMM with frequency measurement would be able to tell you if there's _any_ data on the MIDI out.

The break condition means the input is held in a "space" state for longer than the expected bit time. This usually means one of three things.

1. Wrong baud rate. This is unlikely the case, since MIDI is always 31250
2. Excessive noise. This could be the case, but is also unlikely if the only equipment change is the card to get it to work
3. Wrong pinout/not supported. This is the most likely case. On game ports that don't support MIDI, the same pin is used for the ground return for buttons 3 and 4. The "space" state is logic low, so grounding the pin would result in the break condition.

Higgy69, 2020-11-26:
Hi,
Sorry for delay in response. 
I have USB-to-MIDI and an AMIGA to MIDI cable, but they all have male plugs, i've no sockets on any of my MIDI stuff.
I have a USB logic analyser, but have never used it so would need to read up/learn. 
I updated to latest 7.1 and noticed there was a RS232 connection FAQ, so I soldered some header pins to my Pi-MT32 hat and first off got it working on my Yamaha YMF724 card then swapped to the FM-801AU card, and it works fine :)
I mean to contact people on vogons forum who got the FM-801AU working to see if they have Gameport wiring etc, and later sometime to learn about my Logic Analyser.
On a side note I have bought some rotary encoders ready to hack onto my Pi-MT32 hat for when menu function comes ;) I've not had any luck swapping Soundfonts etc with DOS SysEx commands.
Thanks all

dwhinham, 2020-11-27:
Glad the RS232 page was useful to you!

So does this mean you are connecting the Pi to the COM port on your PC whilst one of these cards is present in the machine?
Or did you just use the electrical interfacing info on the page to inspire a custom wiring job between the Pi's UART and the gameport directly?

Either way, it sounds like you're having some success!

I've attached the DUMP program and a collection of mt32-pi SysEx commands that you can send with it; give these a try with DOS:
[mt32-pi-sysex.zip](https://github.com/dwhinham/mt32-pi/files/5607221/mt32-pi-sysex.zip)

Higgy69, 2020-11-27:
I am connecting the COM port to the Pi while soundcards are in PC. 
Pi-MT32 hat does not have a hole for 3.3V header pin so I soldered one to the exposed GPIO pin, which is right next to the GND and Pin10 header holes.
I am currently using a MAX3232 with integrated 9pin DType. But this is too big to fit in case. So I have just ordered a micro MAX3232 board and a USB A female to DIP.
I plan to use a USB male-male cable and cut off one end and fit a DType. I think only 3 pins are needed. TX RX GND.
With the female USB port added on my Pi-MT32 I was then thinking I could directly connect to MiSTer via UserIO port to avoid using USB-MIDI, but that method I think only works with Cores that have been modified and not from Main Repo?

Higgy69, 2020-12-03:
I don't know if there is a general 'chat' area? But I am getting off topic. I finished my SERIAL COM mod. I got one of those Micro MAX3232 chips and then wired it to a female USB (smaller footprint than 9pin Dtype).
Interesting with MT32 over COM, a cr*ppy Soundblaster Live! card (CT4780) works with SoftMPU!  COM connection give a lot of life :)

I did find some Yamaha XG soundfonts, so I will need to upload the link to the Wiki.

[image]

closed 2020-12-09 by dwhinham