dwhinham/mt32-pi #44: New hardware - clumsyMIDI :)

gmcn42, 2020-11-26:
Hi there!

I had some spare time on my hands and decided to design a GPIO-MIDI+DAC+SSD1306-OLED PCB.
By using the GY-PCM5102 breakout board, I could keep it through-hole only. That's why it's called clumsyMIDI. :)

Assembly is pretty beginner-friendly and components are cheap and easy to come by.
It includes two DIN sockets with MIDI IN and OUT. Hardware MIDI Thru is available as a pin header due to the limited space (can be rewired to MIDI Out socket if that's not used).

I have tested it with mt32-pi's `gpio_thru` option and everything works as it should. Apart from that, MIDI Out is probably most useful for ALSA/Linux right now.

For pictures and more info, take a look at https://github.com/gmcn42/clumsyMIDI
I have 4 more blank PCBs I'd be willing to send away for the price of shipping (from Germany) if somebody is interested in trying it out.

Thanks for your great work on mt32-pi!

dwhinham, 2020-11-26:
Hi!

I've been waiting for someone to come up with something like this - fantastic! Thankyou for sharing this!

I'll update the README shortly with a link to your project - great work! 😃

nswaldman, 2020-11-27:
Neat! This project really pushes a lot of buttons for me :) Would love to get my hands on a pcb to see how idiot-proof it is.

dwhinham, 2020-11-27:
I've now updated the README - I hope the wording is OK! 🙂

closed 2020-11-27 by dwhinham

gmcn42, 2020-11-27:
> I've now updated the README - I hope the wording is OK!

Perfect, thank you!

> Neat! This project really pushes a lot of buttons for me :) Would love to get my hands on a pcb to see how idiot-proof it is.

@nswaldman : Sure thing! Write me via gmcn42[at]gmail.com and we'll arrange that. :+1:

commit 519307b4 (2020-11-27): Add clumsyMIDI to DAC and community hardware lists

Closes #44.