dwhinham/mt32-pi #53: PCM5102 Debugging + Documentation

Purplegopher, 2021-01-04:
Hello there, sorry to bother you.

I have been fighting to get some PCM5102 boards to output anything. I couldn't find a wiring diagram on the wiki for them so I had a look at the other hats.

This seems to be how it is supposed to connect?
1 SCK => GND
2 BCK => Pin 12 (I2S CLK/BCLK)
3 DIN => Pin 40 (I2S DOUT)
4 LCK => Pin 35 (I2S WS/LRCLK/FS)
5 GND => GND
6 VIN => +5V

I bought two and neither work this way (pwm works fine if enabled). Output to i2s is enabled in the config. i2cdetect shows nothing on raspbian (but since there is no i2c control connection, that is probably ok?).

My questions are, are those wiring details correct, are they already in the wiki somewhere? Is there anything I am missing? From the wiki, the PCM5102 should be just "connect and go", or is there something else?

I updated to 0.8.1 and started with a fresh mt32-pi.cfg and only changed the output_device=i2s, no change. I have removed all connected displays and GPIO Midi.

Sorry for the probably stupid thing, it is absoluty a user error, but I simply can not find my error (other than maybe I have killed my PIs and/or the audio boards). Anything I can check?

Purplegopher, 2021-01-04:
OK, I found it....
On the backside of the boards are four solder "jumper". The a-B-c type. My boards had no solder on them, nothing was configured. While I couldn't find a documentation what those jumpers do, I found a picture with soldered pads and replicated that pattern.
Now I have sound!

I'm leaving this open just in case you want to add the wiring diagram and this info to the wiki.

Thank you!! I now have beautiful sound!

Purplegopher, 2021-01-04:
Ah, what the heck, I'll close it, I'm probably the only one that makes this mistake :)
Thank you for your awesome work!

closed 2021-01-04 by Purplegopher

dwhinham, 2021-01-05:
Well done on fixing it yourself so quickly. Actually, you're not alone when it comes to difficulties with these boards!

Users of the clumsyMIDI expansion have encountered this problem: https://github.com/gmcn42/clumsyMIDI/issues/1 

A note has been added to the [clumsyMIDI README](https://github.com/gmcn42/clumsyMIDI#an-important-note-on-the-dac-board-solder-bridges) to inform people to check the solder bridges, which may be set incorrectly, or not at all (as in your case).

The wiki could use a page about this DAC, I'll work on something.
Thanks for testing!

dwhinham, 2021-01-05:
I've added a new wiki page here: https://github.com/dwhinham/mt32-pi/wiki/GY-PCM5102-DAC-module

Purplegopher, 2021-01-05:
Awesome, thank you!!