dwhinham/mt32-pi #54: Implement GM/GS reset SysEx handling

gmcn42, 2021-01-10:
As a little educational project I am currently working on a DOS Real-mode CLI tool that sends mt32-pi control commands (I am basically frankenstein-ing the [DOSMID](http://dosmid.sourceforge.net/) MPU401 code, seems to work so far).

Playing around, I noticed that while mt32-pi in Munt mode - predictably - recognizes an MT-32/LA-Reset SysEx, it does not seem to care about either GM  or GS resets while in Soundfont mode.

MT32/LA Reset (works in MUNT mode): `F0 41 10 16 12 7F 01 F7`
GM Reset: `F0 7E 7F 09 01 F7`
GS Reset: `F0 41 10 42 12 40 00 7F 00 41 F7`

To test: Send a couple hanging note-ons on channels 1-3, then send a reset. The notes will stay on.

It may be nice to implement this as a FluidSynth soft-reset if that's possible. Some MIDI files and games seem to send these at the start. For example `StarGame.mid` has a GS reset in the very beginning.

Unfortunately, in their manuals Roland only specify a 50ms grace period after sending a reset. Don't know if that's a bit too tight for resetting FluidSynth.

dwhinham, 2021-01-10:
Thanks for this suggestion! This will be easy to implement.

I've been thinking about these commands too, as they could be used to (optionally?) automatically switch synths when one of these messages is received.

FluidSynth also has a `synth.midi-bank-select` option to tell it how to interpret MSB/LSB CCs to select instruments, and these should probably be hooked up too.

Just for completion, we also have:
XG Reset: `F0 43 10 4C 00 00 7E 00 F7`

50ms is an eternity for a modern system like the Pi, I don't think we'll have any problems there. 🙂

dwhinham, 2021-01-11:
I've taken a look at this and implemented in the `gm-gs-reset` branch.
There's a test build here that you can try: https://github.com/dwhinham/mt32-pi/actions/runs/476386791

This has nicely fixed some quirks I've noticed when playing several MIDIs in a playlist; e.g. if a song fades out all the channels at the end, the next song would be silent. As long as the next song contains a GM/GS reset command, everything's now good. 🙂

While I was at it, I implemented something that was bugging me. GS synths can use multiple drum channels, so songs that used this feature would sound wrong. This song is a nice example - before, Channel 11 would sound like a piano instead of a triangle:
[mk64_04.zip](https://github.com/dwhinham/mt32-pi/files/5793579/mk64_04.zip)

---

Unfortunately `synth.midi-bank-select` in FluidSynth cannot be switched at runtime; it requires the synth context to be recreated, so the XG reset will need some more thought; it might be possible to modify FluidSynth to allow this. `synth.midi-bank-select` defaults to `gs` though, so GM/GS stuff should be fine at least.

gmcn42, 2021-01-11:
Awesome! I can confirm that the `gm-gs-reset` test build works as expected for me. :)

I guess that having good GS/GM compatibility out-of-the-box is what most people will want, so that's already great and a good default setting, also considering mt32-pi comes with a GS soundfont.
If one wants more than that, I know that FluidSynth has a lot of possible options for Soundfont transposition and stacking and whatnot. Honestly though, I've found this quite messy to configure even in a regular desktop-fluidsynth context. Not sure how much of this complexity is worthwhile to be exposed to users by mt32-pi.

OffTopic: Btw, love the SC-55 style 16x16 graphics output. :) I am looking into adding 1-bit 16x16 BMP loading to my DOS tool, let's see how that goes. :D

closed 2021-01-16 by dwhinham

dwhinham, 2021-01-16:
> Awesome! I can confirm that the `gm-gs-reset` test build works as expected for me. :)

Great - this is now in [v0.8.3](https://github.com/dwhinham/mt32-pi/releases/tag/v0.8.3). 🙂

> If one wants more than that, I know that FluidSynth has a lot of possible options for Soundfont transposition and stacking and whatnot. Honestly though, I've found this quite messy to configure even [...]

Yes; I'll have to think of the best ways to expose the settings that matter the most in a way that's simple and doesn't get too messy. I think the stacking functionality doesn't make much sense for GM/GS/XG purposes because everything should really be self-contained in one font - I think it makes more sense when you have say, a piano-only font and you want to build up a custom set of instruments.

> OffTopic: Btw, love the SC-55 style 16x16 graphics output. :) I am looking into adding 1-bit 16x16 BMP loading to my DOS tool, let's see how that goes. :D

Me too, it's such a gimmick but I couldn't resist implementing it! 🙂 Maybe people will find creative ways to use it with your new utility!

Thanks again!

commit 27042a12 (2021-01-16): Reset FluidSynth on GM/GS reset SysEx

Closes #54.