dwhinham/mt32-pi #58: Support for 16x2 LCD displays

lausvi, 2021-01-21:
First of all: thanks, this is a lovely project :) I am slowly building one!

Suggestion for the LCDs: add support for easy-to-get 16x2 LCD panels. 

I am using one now (config'd as 20x2) so the lower row ends with " |vol " and the volume is not shown. However the four characters at the end of the top row seem to be unused, so couldn't the volume (percentage or whatever) be shown there? Or just replace the |vol with volume in percentage (or as a graph-block like the channel indicators during playback).

dwhinham, 2021-01-23:
> First of all: thanks, this is a lovely project :) I am slowly building one!

Excellent! Feel free to share your build in the Discussions area if you like - and thanks for trying it out 🙂

> Suggestion for the LCDs: add support for easy-to-get 16x2 LCD panels.

I'll certainly consider it; I don't have any 16-char wide screens to test but I think I can just use my 20-char screens.
The first thing I need to do is implement text scrolling so that long messages can be displayed properly.

Thanks!

lausvi, 2021-01-23:
Thanks! I have a breadboard-build for testing, still waiting for some more parts but the basic stuff is working :)

As 16-char is less than the MT32, surely it would be Ok to use abbreviations if scrolling is too much a bother!

lausvi, 2021-02-01:
Added some photos of my designs in the discussion area :)

Probably due to the 'unsupported' 16x2 display I am using, the LCD-reverse parameter didn't work on it.

dwhinham, 2021-02-02:
Thanks - I had a look; great job 🙂

> Probably due to the 'unsupported' 16x2 display I am using, the LCD-reverse parameter didn't work on it.

Do you mean the `rotation = inverted` setting? Unfortunately this is only available for the graphical OLED displays as they have a command to flip the screen. The character LCDs don't have this feature (at least none of the ones I've come across so far).

arkayii, 2021-05-01:
If I may please humbly add my voice to those asking for 16x2 support.

I was hasty enough to do a build using one, foolishly choosing not to check whether or not they were supported until after it was built.

So I will look forward to any minor update enabling support if possible please, or alternatively I will try to make some alterations myself. (in that event, any time-saving pointers to specific code blocks would be appreciated)

The only things I would alter would be to, in MT-32 Mode, remove the text "vol" and instead display there the simple 3 character volume level, and in Soundfont Mode, to left-justify the equalizer graphic. Nothing else greatly matters.

dwhinham, 2021-05-04:
The LCD code is undergoing a major rewrite, which is taking some time, so hacking in any "fixes" for 16-wide at this point would be a bit of a waste of time (and I'd reject any PRs against the currently-released LCD code because of this).

I will aim to include this as part of the next release.

arkayii, 2021-05-04:
Thanks Dale that will be cool. 

Some of us are discussing our amateur mt32-pi (physical) builds at Vogons if you're interested in seeing.
https://www.vogons.org/viewtopic.php?f=29&t=80237

closed 2021-06-25 by dwhinham

dwhinham, 2021-06-26:
Hi @lausvi and @arkayii,

Version [0.10.0](https://github.com/dwhinham/mt32-pi/releases/tag/v0.10.0) is now available and it includes my attempt at supporting 16-character wide displays. In this version, the MT-32 status text is shortened, and longer messages are also now scrolled across the screen.

I don't have a 16-char display myself, so I've only been able to test it on my 20x4 screen by treating it as a 16x2, but hopefully it will work on the real thing.

Please feel free to test it out and hopefully it will work acceptably for you.

arkayii, 2021-07-18:
Finally had a chance to update and check it out. Looks great on the 16x2! Thanks a lot.

arkayii, 2021-07-18:
( trivial note: the mt32-pi.cfg comment line which reads _# Values: 20-128 (128*)_ can now be updated to specify _16-128_ )