dwhinham/mt32-pi #59: Pi4: Only works in "low voltage" scenario

alexoughton, 2021-01-23:
Hello! I really appreciate this project, thank you! I've been working on getting it working with a minimal setup using "mister" and nothing but a USB cable to the User port. I'm finding some very strange behavior where I can only get the mt32-pi to work when the "Low voltage" warning is present. Specifically:

1. Powering the pi4 using the USB cable from user port: "Low voltage" warning is present on virtual OSD, but everything otherwise appears work perfectly. All music seems to work correctly.
2. Powering the pi4 using the USB cable from user port, but with avoid_warnings=2: Virtual OSD shows only garbled noise output, and playing music results in only a few notes playing with incorrect instruments.
3. Powering the pi4 via external USB-C power supply: Same as scenario 2.

So in short it appears that something about the "low voltage" state is the only thing getting things working for me. Suppressing that state or fixing the root-cause with a better PSU causes everything to break-down. Any thoughts here? Perhaps something about the throttling/down-clocking is actually desirable for my system? Is there a parameter I can use to test this further, and maybe deliberately apply it even with a "good" PSU connected?

Thanks in advance!

dwhinham, 2021-01-23:
> Hello! I really appreciate this project, thank you!

My pleasure, thankyou for your interest! 🙂

> I've been working on getting it working with a minimal setup using "mister" and nothing but a USB cable to the User port. I'm finding some very strange behavior where I can only get the mt32-pi to work when the "Low voltage" warning is present. 

In all cases where other users have experienced similar issues, it's been due to the length of the cable used. The problem is that I2C is not really designed for use over a cable like this, and longer cables and/or poor shielding can corrupt the data. If you can shorten the cable <30cm (the shorter the better), these problems should clear up. 30cm is pretty much the absolute maximum anyone's managed to get away with so far.

When `avoid_warnings=2` isn't set and the voltage drops, the Pi's GPU-based firmware starts throttling the CPU, which means it will draw less current. I'm guessing that if there's less current flowing down the USB cable, the I2C data on the adjacent wires is less likely to get corrupted by electromagnetic interference, which would explain things appearing to work properly when the Pi isn't running at full speed.

We're still learning about this as people try different setups, so I'll be updating the docs with cabling recommendations as more test results surface.

You could try removing `fast=true` from the `cmdline.txt` to prevent the Pi from running at full speed, but I wouldn't call this a solution as you'll lose all guarantees on sound quality.

I'm pretty confident that shortening the cable will solve everything, so please do report back if it works for you.

Thanks!

alexoughton, 2021-01-23:
That was, of course, the solution! The cable I was using was somewhere around 3 feet long. I cut it down to just under 30cm, and now it's working perfectly. It makes perfect sense that the increased current draw would result in increased EM interference, but sadly the analog world is not my usual area and so it never occurred to me.

Thanks again for your help with this and for your work on this awesome project.

closed 2021-01-23 by alexoughton

dwhinham, 2021-01-23:
Excellent, thankyou for confirming!

I'm updating the MiSTer user port wiki page right now to try and document this better.

Enjoy! 🙂