dwhinham/mt32-pi #62: USB pinouts dont match

synrais, 2021-01-28:
The wiring for MT32-Pi does not match the hats pdf.

https://github.com/dwhinham/mt32-pi/wiki/MiSTer-FPGA-user-port-connection
https://github.com/MiSTer-devel/Hardware_MiSTer/blob/master/releases/MT32Pi_Main.pdf

https://imgur.com/3KY8dnk

synrais, 2021-01-28:
Looks like the diagrams Ts are swapped with the Rs

synrais, 2021-01-28:
Well it seems it has changed.
This is the current working wiring for mixing audio back into the MiSTer.
https://imgur.com/rf5lna5
Orange must be left N/C if its connected to 8 nothing works.

dwhinham, 2021-01-28:
You haven't mentioned anything about how you are constructing your cable. Did you cut a USB A-B cable?
The TX/RX signals names are swapped depending on your point of view:

[image]

> Orange must be left N/C if its connected to 8 nothing works.

The wiki's pinout does not tell you to connect to pin 8 of the Pi. Though it should do no harm; it would just be used to send MIDI data to the MiSTer, though this is unused right now.

> That will be an issue for people using USB2USB cables though i'm sure.

Pin assignment has not been an issue for anyone using A-A cables and the official HAT so far, I'm not sure what you mean here.

synrais, 2021-01-28:
Maybe for the pin out diagram change the colors to half half on RX TX to indicate those ones are crossovers, it could help simplify things.

dwhinham, 2021-01-28:
I've reworked that section of the wiki page to try and clarify things - let me know what you think.

closed 2021-01-30 by dwhinham