dwhinham/mt32-pi #66: Button to Switch rom does snot cycle MT-32 New and Old

Zhan85, 2021-02-02:
As the title says.  When wiring a button to button 2 it will switch between CM32L and whichever MT32 Rom is default at startup but not between either new or old.

dwhinham, 2021-02-02:
A "new" ROM is a v2.04 ROM. It sounds like you don't have it present. Please check the wiki page and verify what ROMs you have: https://github.com/dwhinham/mt32-pi/wiki/MT-32-synthesis#rom-support

Zhan85, 2021-02-02:
Ok thank you!  Not sure if I am allowed to ask you this but does that New rom actually exist to download anywhere?  I have never come across it in my searches.

dwhinham, 2021-02-02:
No problem 🙂

You're right, the 2.04 ROM is not available for download online anywhere, and I've searched very hard. In the end, I dumped the ROM from my real MT-32.

You aren't really missing anything without the v2.04 ROM; for Munt it is functionally identical to the CM-32L ROM, but without the extra sound effects. It doesn't really offer any benefits other than for "completeness" or a curiosity.

Obviously I can't discuss sharing copyrighted material here.
Thanks!

closed 2021-02-02 by dwhinham