dwhinham/mt32-pi #70: Oled flickering on and off under 8.4

Xargon321, 2021-02-09:
Tried upgrading to 8.4 today and my oled started flickering on and off once you started to play music it would stay on but i think it was trying to turn off but every time a note hit it would come back on made it look like it was refreshing weird 

I tried replacing all the files and just replacing kernel8 same thing as soon as i roll back to 8.3 it works as is should

Its a rbi3+ and using ssd1306_i2c oled screen

dwhinham, 2021-02-09:
Please post full details of your screen and where you got it from, and also attach your config file.

Thanks!

Xargon321, 2021-02-09:
https://www.amazon.com/gp/product/B08L7QW7SR/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1

that's the ones i got and my config file is here

#          __ ________  ________                  .__
#   ______/  |\_____  \ \_____  \          ______ |__|
#  /     \   __\_(__  <  /  ____/   ______ \____ \|  |
# |  Y Y  \  | /       \/       \  /_____/ |  |_> >  |
# |__|_|  /__|/______  /\_______ \         |   __/|__|
#       \/           \/         \/         |__|
#
# mt32-pi.cfg: mt32-pi configuration file.

# Default options are marked with an asterisk (*).

# -----------------------------------------------------------------------------
# System options
# -----------------------------------------------------------------------------
[system]

# Set the default synthesizer to be made active on startup.
#
# If the default synthesizer is unavilable (e.g. missing ROMs or SoundFonts),
# the first working synth is made active.
#
# Values: mt32*, soundfont
#
# mt32:      Use mt32emu (Munt) for Roland MT-32 emulation
# soundfont: Use FluidSynth for SoundFont synthesis
default_synth = mt32

# Set the I2C baud rate/clock speed for all peripherals (Hz).
#
# Most peripherals will work fine at the default speed (400KHz "fast mode"),
# but larger LCD/OLED displays (e.g. 4-line I2C HD44780 and 64 pixel high
# SSD1306) won't be able to refresh at 60FPS at the default setting.
#
# Try increasing this value to 1000000 (1MHz) for a smoother LCD refresh rate.
# If your display doesn't work, try backing off the speed 100KHz at a time
# until it does.
#
# Values: 100000-1000000 (400000*)
i2c_baud_rate = 400000

# Set the timout for power saving mode (seconds).
#
# After the specified number of seconds of silence, the CPU clock speed will be
# reduced, the audio device will be stopped, and and the LCD's backlight will
# be turned off to save energy (certain I2C displays only).
#
# Any MIDI activity will instantly bring the system out of power saving mode.
#
# If set to 0, power saving mode is disabled.
#
# Values: 0-3600 (300*)
power_save_timeout = 300

# -----------------------------------------------------------------------------
# MIDI options
# -----------------------------------------------------------------------------
[midi]

# Enable or disable searching for a USB MIDI interface on startup.
#
# Disable this to speed up boot time if you are using GPIO for MIDI.
# If no USB MIDI devices are detected, GPIO will be used as a fallback.
#
# Values: on*, off
usb = off

[... 228 lines cut ...]

# Set the height of the LCD.
#
# Same characters vs. pixels considerations as for width.
#
# Values: 2-64 (32*)
height = 32

# Set address (hexadecimal) of I2C LCD.
#
# This will be used to communicate with LCDs connected via the I2C bus.
#
# Values: 00-80 (3c*)
i2c_lcd_address = 3c

# Rotate the display output (graphical LCDs only).
#
# Some graphical displays support rotation. Use this option if you need to turn
# the display around.
#
# Values: normal*, inverted
#
# normal:   No rotation
# inverted: The display output is upside down
rotation = normal

dwhinham, 2021-02-09:
Thanks for the info, your config was of course fine.

I was able to quickly reproduce it here, sadly I broke 128x32 screens in the last version - sorry about that!

Please test this build here: https://github.com/dwhinham/mt32-pi/actions/runs/551181587
You can just download the "kernels" archive and replce the kernels.

Xargon321, 2021-02-09:
all working like it should

commit b38fa820 (2021-02-09): Fix incorrect SSD1306 framebuffer packet size

Fixes #70.
closed 2021-02-10 by dwhinham