dwhinham/mt32-pi #103: v0.9.1 not booting

digitaldiatribe, 2021-03-22:
Howdy. I updated from v0.8.5 and after doing so, it does not boot up and buttons on the pi-hat do not respond. I went back to v0.8.5 and it booted right up. I tried doing a fresh format on the micro SD card and tried v0.9.1 again but still would not boot. Pi used is a 3 A+.

I've gone back to v0.8.5 for now.

dwhinham, 2021-03-22:
Assuming the MiSTer HAT as you didn't say. It sounds like you didn't follow all of the setup instructions to enable the screen and buttons.

If you didn't edit `mt32-pi.cfg` at any stage then you have missed the most important part.

Please follow the steps in the README carefully and try again.

closed 2021-03-22 by dwhinham

digitaldiatribe, 2021-03-22:
Whoops, I totally forgot about that file. I just copied and pasted the contents over the original install. Sorry about that.

dwhinham, 2021-03-22:
No worries, enjoy! 🙂