dwhinham/mt32-pi #104: Rotatory Encode Reverse Option

shaeon, 2021-04-06:
Hello I'm using Edu Arana PI-MIDI 1.3:

[https://www.arananet.net/pedidos/product/pi-midi-1-3-a-baremetal-mt32-emulator-using-raspberry-pi-preorder/](url)

The problem its that here rotary encoder its working reverse, if you turn left up volume and if you turn right down volume.

I try find a option for reverse this issue but i can't find. 

Can you add an option for can put reverse working in next version?

Thanks so much for your work!

dwhinham, 2021-04-06:
Hi!

I can add this as an option, but we should also inform @arananet so that he can check the schematic / encoder.

@arananet: the prototype Pi-MIDI 1.3 that you sent me works with the correct direction - did anything change in the encoder wiring between the prototype and the final version?

If not, did you use a different part for the encoder? It may be a good idea to check this.

Thanks.

arananet, 2021-04-06:
Hi Dale,

No, is the same as you got there, but it must be related to the actual encoder itself. This is the schematic.

[image]

shaeon, 2021-04-06:
https://user-images.githubusercontent.com/10370710/113720410-e26b5580-96ee-11eb-86ac-2e4067541fc3.mp4

I upload a video working with encoder.

dwhinham, 2021-04-06:
Thanks both,

I guess the encoder must have a different pinout which is unfortunate, but not a huge problem - I can just ad a config option to negate the direction.

Will be added in the next version.
Thanks!

arananet, 2021-04-07:
Thanks Dale!.

dwhinham, 2021-04-07:
There is now a test build available here: https://github.com/dwhinham/mt32-pi/actions/runs/727192835

A new config option is available:
```ini
[control]

# ...

# Reverse the rotary encoder direction (if used by control scheme).
#
# Some rotary encoders may have their CLK/DAT signals swapped, resulting in
# a reversed rotation direction.
#
# Use this option to correct the direction.
#
# Values: on, off*
encoder_reversed = off
```

You can just download the `kernels` artifact and edit your config to test it.

Thanks!

shaeon, 2021-04-08:
Wow! So fast Dale!

It's work as expected. I download new kernel and put on encoder_reversed option in control and now my brain can be relax :)

closed 2021-04-10 by shaeon

dwhinham, 2021-04-10:
Great! Thanks for testing, it'll be included in the next release. 👍

commit eec83451 (2021-04-07): Add configuration option to reverse rotary encoder

Closes #104.
commit a438a129 (2021-05-01): Add configuration option to reverse rotary encoder

Closes #104.
commit c7f4ca99 (2021-06-14): Add configuration option to reverse rotary encoder

Closes #104.
commit 2415b7b2 (2021-06-25): Add configuration option to reverse rotary encoder

Closes #104.