dwhinham/mt32-pi #106: USB gadget mode

fayep, 2021-04-07:
I just got the MiSTer PI hat and it is great in regular cores but it does not appear in eg. ScummVM or Basilisk. If I could plug a USB cable from the MiSTer USB (not user) port to the Pi directly and have it detected as a MIDI class device directly using USB gadget mode that would be amazing. It seems like overkill to need to purchase and attach 2 USB to MIDI adaptors, one to the Mister and one to the Pi so that they can exchange data. I think this would also render the the MT32 PI instantly usable with other devices compatible with USB midi. 
https://github.com/atomic-penguin/linux/blob/master/drivers/usb/gadget/f_midi.c might be a good starting place?

dwhinham, 2021-04-07:
The real place to request this would be rsta2/Circle, and it was brought up already here https://github.com/rsta2/circle/issues/11 and here https://github.com/rsta2/circle/issues/202

I agree this would be a nice idea, but I just don't have the skills/experience to do it and it would be an enormous amount of work - the RPi USB driver has no public documentation so we're pretty lucky to have it working at all under baremetal.

As a workaround for the MiSTer use case I do have network MIDI almost working, so with a little bit of extra software on the MiSTer Linux side it might be possible to do things that way.

fayep, 2021-04-07:
Only if you are using a PI B or plug a USB ethernet in though?  I will take a look at this at the weekend and see how much work is involved. Thanks for the pointer to Circle

dwhinham, 2021-04-07:
No, it'll support Wi-Fi too - I've actually had some very good results so far in my testing with Wi-Fi re: latency, much better than I expected.

No problem - if Circle did end up picking up gadget mode support I'd jump on supporting it here too, of course; would be a nice touch.

fayep, 2021-04-08:
So I've found this http://www.isticktoit.net/?p=1383, so at least we know there's stuff working with relatively modern kernels.  That should make things a little easier, hopefully.

c0d3h4x0r, 2022-02-19:
Any more information on this?  I would love to be able to use my MiSTer mt32-pi with my PC or RetroPie via just a USB cable.