dwhinham/mt32-pi #107: Pi 3 B 1.2  (+)  issues

ppuskari, 2021-04-09:
Was really excited to try this project out.  So far I've tried with a M-Audio Uno and a generic.  The Generic adaptor actually works better when it works.

I have basically given up on this for a while I think.  I normally run FluidSynth on Linux and Windows and the same HT-700 keyboard and both adaptors work great with that.

On this project even with a 5 second delay, most of the time the adaptors are not found, or when they do and I see the activity lights, there is no sound off the pwm headphone jack.

Sometimes I can get the system to start and produce sound by moving the usb port,  When I do get that working then the audio quality is horrible, and on instrument changes it hangs the sound and eventually the system locks up it seems and I have to reboot.

Been using the Raspi since 2012 with midi devices and timidity, fluidsynth etc and those work well.  I have all the verfied sha-1 roms and pcm instrument data files except the NEW mode one 2.04   Can't find that one at all on the web.

I have found that only really small soundfonts work too.  The included one and the 8meg soundblaster one mainly, but they suffer form the hanging sound and crashing too.

Not sure if I'm expecting too much from this project.   Am considering building munt on the pi ontop of normal raspbian and see if that works

Have also tried with 2 different 3 b+ pis with the same results.

Any way to get at some logging to see what the system is even struggling with?

dwhinham, 2021-04-09:
FIrst let's start with the basics - what kind of power supply are you using?

If the PSU is weak, the Pi will be throttled and you will get severe audio problems. Make sure your PSU can deliver at least 2 amps.

If your Pi shows a lightning bolt on screen (HDMI) when you boot Linux on it, the PSU is not suitable.

Regarding USB MIDI adaptors, in general they aren't as compatible as they are under Linux, mainly because the Pi's USB controller lacks public documentation, so it has been difficult for the Circle author to get every device working. TL;DR, USB is hard.

That said, the two adaptors you've mentioned have been the most problematic for mt32-pi users so far - I have an old M-Audio Uno that usually works fine, others have used more recent revisions and they have been unreliable. The best results have been with Roland adaptors or using a GPIO MIDI interface.

You aren't expecting too much from the project - most users enjoy a very good experience and I personally use it every day. It wouldn't be published here at all if it was completely unusable 🙂

If you want to see the logs, you can try the `kernels-hdmi` download from this build, which is from a recent commit: https://github.com/dwhinham/mt32-pi/actions/runs/727192835

Just replace the kernels and attach an HDMI display, you should see the logs.

My guess is that you may see Low voltage warnings and/or USB errors. Let me know how you get on.

andrewwong2000, 2021-04-10:
I've been doing some bench testing using a good digital lab power supply.. if I run a DE10+Analog+Pi3A+MT32pi,  I don't see more than about 1.4A or 7.2W total draw. So I'm not sure 2.4A is the problem.

I tried a 3A+ and a 3B+ which both work fine if you power the Pi separately, but trying to run off the same PS as the DE10 leads to errors, audio glitches (with low voltage warning). 3B+ doesn't even boot properly.

Seems that the DE10 on board regulator is going to limit power supply to the Pi.. might work for some people but not all.

dwhinham, 2021-04-10:
Thanks for the testing reports, but the original poster did not mention MiSTer or DE-10 at all, so let's assume they are using mt32-pi standalone to avoid confusion.

ppuskari, 2021-04-10:
Hello!  Thanks for looking into this. Yes this is a stick 3b and 3b+.  The first was a canakit 2.5amp PS then also tested with another 2.5amp supply from Fryes.  The cans kit one however did throw undervolts sporadically.  The Fryes one dies not.

Did more testing with both boards with the hdmi kernel.

Loghinh shiws when most thjngs are working that the last log is cpu 0 starting main task.  This is regardless of mt32 or sound font mode.

Sound font mode however will at least generate sound before locking up on instrument changes. But I can't rule out its either USB dongle because it will throw invalid bytes according to the logging.

I still do not get any sound in mt32 mode. Yet.  Will try hooking up some gpio midi next and see if that helps the first issue at least.

Is there supposed to be some kind of logging when it finds mt32 roms like I see for soundfont?  I have sha1 matched roms in the rom folder.  And using default old mode when I try.

Yes this is bare pi with no mistr integration or add on hats. Just the USB dongle.

Thanks!

dwhinham, 2021-04-10:
There's no logging for MT-32 ROM scanning - if there's at least one valid set it'll work, otherwise it will fail and you'll only have SoundFont mode.

dwhinham, 2021-04-10:
Can you please [read all of the FAQ](https://github.com/dwhinham/mt32-pi/wiki/FAQ#common-problems-and-solutions) aswell, because it sounds like you might be trying to play MT-32 mode on Channel 1, which of course will not work.

ppuskari, 2021-04-10:
That is correct I have the midi keyboard a Casio ht700 on channel 2 for transmit. It works great on that channel normally as well.

In soundfont mode it keys the HD light upon midi send data. On mt32 mode it also lights up the pi HD light on keysend data on mt32 mode as well. Just no sound at all out the pwm headphone jack

I'll keep plugging away. Yes l have been through the faq a lot trying to see if I just missed something like that. But no.    It did take a while to find the roms properly as well as interleave properly a few of them from smaller eprom pulls. But everything matches according to your sha1 computes for each file I'm using. 

Layer this afternoon I'll build the din5 to gpio cable and try that method.

closed 2021-04-30 by dwhinham