dwhinham/mt32-pi #138: 💡 [FEATURE] Add third option to synth select sysex for just passthrough

netham45, 2021-05-25:
I have a dreamblaster and I would like to be able to switch between that and the MT-32. Would it be possible to have a third option for the synthesizer sysex call that enables passthrough to the TX pin and doesn't do any internal audio processing, then when switched away it changes the passthrough setting back to whatever is defined in the mt32-pi.cfg file? Or a sysex to directly control passthrough and an option for disabling the mt32-pi synth?

dwhinham, 2021-05-28:
This would be possible as part of a planned MIDI routing feature where all different kinds of MIDI inputs (GPIO, USB, Pisound, network) can have their inputs optionally routed to the synth and each other, which will be more flexible.

Basically, `gpio_thru` is going to be deprecated and replaced with something like:

```ini
[midi]
# GPIO input sent to synths and thru to output
gpio_route = synth, gpio

# GPIO input just sent thru to output
gpio_route = gpio

# USB input sent to synth and thru to USB output
usb_route = synth, gpio
```

This would also be controllable via SysEx.

I won't be accepting any PRs for this as I already have a design in mind and the MIDI code is going to be changed extensively.

netham45, 2021-05-28:
>I won't be accepting any PRs for this as I already have a design in mind and the MIDI code is going to be changed extensively.

Heh, you can close my PR request then. Just submitted it before I saw this response.

skadarnold, 2022-07-04:
Looking forward to this being further implemented. I can think of tons of uses for this, even outside of retro gaming. Thank you.