dwhinham/mt32-pi #167: 🐛 [BUG]

ottj, 2021-07-26:
Another entry for the cable wiki:
The right and left angle USB 3.0 adaptors from Delock sadly do not work (pi does not boot properly/ no image on the OLED)

Name:
Delock 65341 and Delock 65342

Length:
N / A

Comment:
No image on screen, no connection to Mister, flashing of LEDs

dwhinham, 2021-07-26:
Thanks, I've added them.

Are you saying that using either of these adaptors combined with another cable causes failure? Does everything work alright with just a cable?

ottj, 2021-07-26:
Yes, with only the cable (that I got together with the mt32-pi) from the Mister to the MT-32 everything works. With one or two adaptors added to the chain it stops working. The OLED stays dark and the mt32-pi is not detected by the mister.

dwhinham, 2021-07-26:
Thanks for the clarification, yes, that sounds like the typical I2C corruption.

I've tweaked the wording of the wiki entry. Thanks for the report.

closed 2021-07-26 by dwhinham