dwhinham/mt32-pi #181: 💡 [FEATURE] FluidSynth commands file

moa3, 2021-09-02:
Hello, I've started using mt32-pi on a Pi 3 attached to a Native Instruments Komplete Kontrol master keyboard and so far, I love it. 
Best thing is, it just works \o/

Now, on the keyboard I have, there is a bunch of rotary encoders that send MIDI CC signals. I have been able to configure them in FluidSynth on my PC by using router commands (as per the manual https://github.com/FluidSynth/fluidsynth/wiki/UserManual)

for example:

```
router_begin cc
router_par1 14 14 0 7
router_end
```

will successfully bind the first rotary encoder to control the master volume in FluidSynth.

I understand that these commands can be set either directly on the standard input, via the network when FluidSynth is started with the `-s` option or via a file passed using the `-f` option.

I have tried a few things to try and run these commands in mt32-pi but have failed so far.

What are your thoughts about adding a way to pass FluidSynth commands ?

Off the top of my head, I can think of at least 2 options. 

 * adding an option in `mt32-pi.cfg` to start FluidSynth in server mode, maybe with a custom port ?
 * load a `fluidsynth.conf` (or whatever better naming you might think of) file when starting FluidSynth if the file exists on the SD card, (or just load it all the time even if it is just an empty file ?)

Thanks a lot for the great work :pray:

dwhinham, 2021-09-02:
Hi there, thanks for the suggestion, I agree that exposing more of FluidSynth's features would be nice.

Currently mt32-pi only makes use of FluidSynth's core synth engine. FluidSynth also has other components that make up its "front end", such as a command interpreter ("shell") - which is what you are accessing by typing these commands, a MIDI router (the part that allows you to remap your CCs), and even a SMF MIDI file player.

Unfortunately a lot of this code depends on Linux threads, which we don't have on baremetal, so for the sake of getting the essential parts working that we care about the most (the sound generator itself), most of this is disabled and not built into mt32-pi.

However, at a glance, the shell component and router look fairly reasonable and it may be possible to port these parts too with some effort. I will need time to think about this and see how feasible it is.

If I got the shell component working, I could probably just replace (or add) the `[fluidsynth]` section of `mt32-pi.cfg` file with FluidSynth's own options and feed lines within that section to the shell component.

The server/remote stuff is technically possible, but will need new code writing, because again FluidSynth's networking stuff depend on Linux-isms and it's not just a straight recompile for baremetal. I think this would be a pretty low priority.

moa3, 2021-09-02:
Hi and thanks for answering my request so quickly.

To be honest, the server part was a bit of an afterthought and I don't think I would really use it much unless I am trying out some conf so, having "just" the shell would be a great addition and more than enough for my use.

So, whenever you have the time and will I guess :-) I'll be happy to do some testing on my hardware if that helps and will certainly make a donation on https://ko-fi.com/d0pefish for support and fuel :coffee: 

Cheers

dwhinham, 2021-09-03:
Thankyou for the donation, I appreciate it!

This is on my to-do list to investigate for a (hopefully not too distant) future version.