dwhinham/mt32-pi #214: Problem with ssd1306_i2c in pi zero 2🐛 [BUG]

antmerino, 2021-12-10:
I use pi zero 2 with a ssd1306_i2c lcd with the compiled version in the develop branch.
When I've been playing something for about 10 minutes, the lcd starts scrolling right and up.
Attached configuration file and a video with the problem.
Thanks

- **mt32-pi version:** Branch develop
- **Raspberry Pi model:** _Raspberry pi zero 2_
- **HATs:** _Pi-MIDI with ssd1306_i2c _
- **USB devices:** _None_
- **MIDI host:** _Mistert_

## Hardware and software

- **mt32-pi version:** _e.g. v0.10.0_
- **Raspberry Pi model:** _e.g. Raspberry Pi 3 Model B+_
- **HATs:** _e.g. Pi-MIDI_
- **USB devices:** _e.g. None_
- **MIDI host:** _e.g. 486 PC via Sound Blaster gameport_

## Bug description

_Provide a clear and concise description of what the bug is._

## Steps to reproduce

_Provide clear steps to reproduce the problem. For example:_

1. Load Monkey Island 2 from MS-DOS.
2. Visit the mapmaker's room.
3. Hear distortion in audio.

## Expected behavior

_A clear and concise description of the behavior you expected._

## Configuration file

<details>
<summary>Click to expand</summary>

```ini
Paste the contents of your mt32-pi.cfg here.
```
</details>

## Videos/audio recordings

_If appropriate, add a short video or audio recording to help explain your problem._

## Additional information

_Add any other information about the context of the problem here._
[mt32-pi.cfg.txt](https://github.com/dwhinham/mt32-pi/files/7689789/mt32-pi.cfg.txt)
![8181643d-84e2-4fbd-8999-29559151f50b](https://user-images.githubusercontent.com/84129844/145509674-1a3a3f6c-ca92-4190-915c-d4d4c3f83

https://user-images.githubusercontent.com/84129844/145509945-a6c2e740-c6a5-492e-97c3-9e5668a326c2.mp4

825.jpeg)

dwhinham, 2021-12-10:
From the [FAQ](https://github.com/dwhinham/mt32-pi/wiki/FAQ#q-im-using-a-mister-interface-pcb-with-a-usb-a-a-cable-and-the-oled-display-is-scrambledthe-osd-controls-dont-workaudio-is-distorted-or-missing---whats-wrong):

> **Q:** I'm using a MiSTer interface PCB with a USB A-A cable and the OLED display is scrambled/the OSD controls don't work/audio is distorted or missing - what's wrong?

> **A:** Your cable is probably too long and the I²C and/or I²S signals are being degraded. Please try a short cable as explained [in this section of the MiSTer wiki page](https://github.com/dwhinham/mt32-pi/wiki/MiSTer-FPGA-user-port-connection#-important-cabling-considerations).

closed 2021-12-10 by dwhinham

antmerino, 2021-12-10:
my cable is https://www.mycablemart.com/store/cart.php?m=product_detail&p=7270 , This's recomend in the wiki and only has 15 cm
we have tested with a raspberry 3 and with other lcd. and this works. 
Please, can you test with your pi zero 2 and this lcd?
thanks

dwhinham, 2021-12-10:
There are no issues with the OLED driver on my Zero 2 W test setup.

You have a hardware issue. Talk to whoever sold you it.