dwhinham/mt32-pi #239: Include Dexed in MT32-PI

mwedmark, 2022-02-08:
Would be really great if the open-source DX7 emulation synth Dexed was included or simply to include in MT32-PI.
What is the thoughs around adding more nice synths?

dwhinham, 2022-02-08:
I've thought about this for a long time, since early last year, as I have a real DX7-IIFD and love it a lot.

If there's demand for this, I'll certainly do it.

I'm fully aware of a porting effort ongoing on the Circle repo ("MiniDexed"), but I'm not sure if their efforts will fit well into mt32-pi's codebase.

Will leave this open until I get time to look into it.

mwedmark, 2022-02-08:
Thanks for the quick feedback! I'll currently don't own a MT32-PI but I'll build one just for fun.  I did grow up with AWE32 and it's fantastic Soundfonts. Pro stuff for a dime. Nice to see the tech working here as well!

human222, 2022-02-14:
Dexed would be fantastic. Imagine composing DX-7 music with one of the computer cores on the MiSTer?

Also as a side-note, MAME can emulate a number of synthesizers, including some FM ones that have a bit of a similarity to the DX line. Just adding this in case the Dexed codebase turns out to be difficult.

dekkit, 2022-02-15:
Just dropped in to say exellent project and daaang that would be aweeeesome to have DX7/dexed!

krulf, 2022-02-15:
Sounds really good if it could be made to work!

dwhinham, 2022-02-15:
> Dexed would be fantastic. Imagine composing DX7 music with one of the computer cores on the MiSTer + mt32-pi?

:slightly_smiling_face:

> Also as a side-note, MAME can emulate a number of synthesizers, including some FM ones that have a bit of a similarity to the DX line. Just adding this in case the Dexed codebase turns out to be difficult.

Good point, I think ultimately MAME's synthesizers would be the better option as they are going for accuracy and complete emulation of the machine. There was some Yamaha MU-series related research going on too, IIRC. I'll look into this.

As for Dexed - the upstream codebase is pretty hard to wrap my head around and the code itself is pretty thin on the documentation.

I took a look at [Synth_Dexed](https://codeberg.org/dcoredump/Synth_Dexed), whose author has done the hard work of providing a sane API for implementing Dexed's synth engine into other projects; this is what MiniDexed is also using.

Right now, I have it compiled-in to mt32-pi, and generating sound from within mt32-pi already, but MIDI keys don't seem to correspond to proper tones. It sounds like FM, but perhaps all the instrument settings are not initialized properly; probably something I'm doing wrong, not sure yet.

Will provide more info later.

human222, 2022-02-15:
>Good point, I think ultimately MAME's synthesizers would be the better option as they are going for accuracy and complete emulation of the machine. There was some Yamaha MU-series related research going on too, IIRC. I'll look into this.

Yeah there's lots of goodies in MAME. I think they started on SC-55 as well, although that one isn't showing life yet. About the MU-series, I tested it myself a while back. It's not perfect but it's quite pleasant already.

[image]

In the screenshot I got one instance of MAME sending MIDI OUT to a loopback device, and then a second instance of MAME receiving MIDI IN from the same loopback device. The loopback device can of course be skipped when using hardware.

>Right now, I have it compiled-in to mt32-pi, and generating sound from within mt32-pi already, but MIDI keys don't seem to correspond to proper tones. It sounds like FM, but perhaps all the instrument settings are not initialized properly; probably something I'm doing wrong, not sure yet.

>Will provide more info later.

That's great too. Well, whichever route you take, it looks like something very exciting is coming up for the mt32-pi. I can't wait to see what happens! :)

probonopd, 2022-02-23:
> I'm fully aware of a porting effort ongoing on the Circle repo ("MiniDexed"), but I'm not sure if their efforts will fit well into mt32-pi's codebase.

Happy to collaborate.

If you find something in MiniDexed that you think is not a good architecture decision, then we'd be more than happy to hear from you.

Metacrack, 2023-03-27:
Why not implementing NukedOPL for a OPL emulation?

farvardin, 2024-09-25:
Having Dexed included into mt32pi would be great, but it's true minidexed can be installed into an other SD card, and it runs fine with a mt32pi-hat, including midi DIN5 and LCD, so I'm quite happy with this setting...

probonopd, 2024-09-25:
Maybe we could find some way to allow for "multibooting" (i.e., selecting MT32-PI vs. MiniDexed vs. other OSes) at boot time?