dwhinham/mt32-pi #240: FTP server always rejects user credentials 🐛 [BUG]

c0d3h4x0r, 2022-02-08:
## Hardware and software

- **mt32-pi version:** 0.11.0
- **Raspberry Pi model:** Raspberry Pi 3 Model A+
- **HATs:** MiSTer mt32-pi hat
- **USB devices:** None
- **MIDI host:** MiSTer USER port
- **Powered via:** MiSTer USER port

## Bug description

I've enabled wi-fi and configured wpa_supplicant.conf, and verified the host `mt32-pi` appears on my home network.
I fire up FileZilla client on my Windows 10 PC and define a new stored connection per the instructions in the mt32-pi wiki.
Then I try to connect... and the FTP server always rejects the login with the following:

```
Command:	USER mt32-pi
Response:	331 Password required for 'mt32-pi'.
Command:	PASS ********
Response:	530 Not logged in.
Error:	Critical error: Could not connect to server
```

I've triple-checked that the username and password are correct as I configured them in `mt32-pi.cfg`.
I've also tried an FTP client on my iPhone, which hits the same error (credentials rejected).

The username is `mt32-pi` (default) and the password is `p4ssw0rd` (not the default).

## Steps to reproduce

See preceding section.

## Expected behavior

FTP server should accept valid user credentials and log the FTP client on successfully.

## Configuration file

<details>
<summary>Click to expand</summary>

```ini
#                                  ________    _______                      __
#                         __      /_____   `. /____   `.                   /__`
#     ____   ____     __/  /____  _______)  / ______)  / ____  ______      __
#   /   __ v  __  `./__   _____//_____    < /   _____. /____//   ___  `. /  /
#  /  /  /  /  /  /   /  /____  _______)  //  /______       /  /____/  //  .__
# /__/  /__/  /__/    \______//_________. /_________/      /   ______.  \____/
# /////////////////////////////////////////////////////// /  / //// /// // /
#                                                         ```
# mt32-pi.cfg: mt32-pi configuration file.
# Default options are marked with an asterisk (*).

# -----------------------------------------------------------------------------
# System options
# -----------------------------------------------------------------------------
[system]

# Enable or disable verbose startup and error output.
#
# When enabled, outputs more information to the LCD when starting up, and when
# MIDI/UART errors are detected.
#
# This also may hide the boot logo on smaller graphical displays.
#
# Values: on, off*
verbose = off

# Set the default synthesizer to be made active on startup.
#
[... 452 lines cut ...]
# This FTP server is a very basic implementation which DOES NOT feature any kind
# of transport layer security/encryption. Therefore, you should NOT enable this
# feature on a public network or expose the Raspberry Pi to the Internet.
#
# The FTP server should be considered a convenience feature only, for performing
# updates and configuration changes without having to replace the SD card.
#
# Values: on, off*
ftp = on

# Set the FTP server username and password.
#
# Values: any ASCII string (*mt32-pi)
ftp_username = mt32-pi
ftp_password = p4ssw0rd
```
</details>

## Videos/audio recordings

N/A

## Additional information

N/A

dwhinham, 2022-02-08:
This looks like a duplicate of #232, could you check [this post](https://github.com/dwhinham/mt32-pi/discussions/232#discussioncomment-1997317) and try again?

c0d3h4x0r, 2022-02-09:
Sure, I'll give that a try.  Thanks for the quick turnaround!

c0d3h4x0r, 2022-02-09:
That solved it.  Thanks!

closed 2022-02-09 by dwhinham