dwhinham/mt32-pi #243: Confirm midi in iRig 2 works and also

lucjanssen, 2022-02-27:
Hi, https://www.thomann.de/nl/ik_multimedia_irig_midi_2.htm midi in works.

also managed to work https://www.amazon.nl/gp/product/B07PMGGMJF/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1
DollaTek PCM5102A, Out of the box with FLT, DMP, SCL and FMT set to ground and XMT to 3,3 Volt on the rpi.

lucjanssen, 2022-03-01:
And of course 

BCK | 12 |  
-- | -- | --
DIN | 40 |  
LCK | 35 |  
GND | 6 |  
VIN | 2 |

dwhinham, 2022-03-01:
This DAC seems to be the common CJMCU-5102 module. I've added a new page to the wiki about this DAC: https://github.com/dwhinham/mt32-pi/wiki/CJMCU-5102-DAC-module

I've also added the iRig MIDI 2 to the tested USB MIDI interfaces list: https://github.com/dwhinham/mt32-pi/wiki/USB-MIDI-interfaces

Thanks for the reports!

closed 2022-03-01 by dwhinham