dwhinham/mt32-pi #247: 🐛 [BUG] Loud noise when switching FluidSynth soundfonts

c0d3h4x0r, 2022-03-02:
## Hardware and software

- **mt32-pi version:** v0.10.0
- **Raspberry Pi model:** Raspberry Pi 3 Model A+
- **HATs:** MiSTer mt32-pi
- **USB devices:** None
- **MIDI host:** MiSTer USER I/O port

## Bug description

If you use the MiSTer OSD to change the SoundFont while the mt32-pi is playing music (over HDMI), there's a short, LOUD blast of noise that sounds like residual buffered audio data leftover from the previous SoundFont.

## Steps to reproduce

1. Load the ao486 core in MiSTer with mt32-pi attached via (and powered by) USER I/O port and configured via the core's OSD to use FluidSynth mode.  Make sure your mt32-pi has multiple different SoundFonts in place to choose from.
2. Using an MS-DOS 6.22 or 7.10 VHD, run any game (such as DOOM) configured to use General MIDI (port 330h) for the music.
3. Let the demo start playing so that background music is playing on the mt32-pi.
4. While the music is playing, use the core's OSD to change to the next SoundFont.

## Expected behavior

No loud garbage noise when switching among SoundFonts or when switching back and forth between Munt and FluidSynth modes.

## Configuration file

<details>
<

```ini
#                                  ________    _______                      __
#                         __      /_____   `. /____   `.                   /__`
#     ____   ____     __/  /____  _______)  / ______)  / ____  ______      __
#   /   __ v  __  `./__   _____//_____    < /   _____. /____//   ___  `. /  /
#  /  /  /  /  /  /   /  /____  _______)  //  /______       /  /____/  //  .__
# /__/  /__/  /__/    \______//_________. /_________/      /   ______.  \____/
# /////////////////////////////////////////////////////// /  / //// /// // /
#                                                         ```
# mt32-pi.cfg: mt32-pi configuration file.
# Default options are marked with an asterisk (*).

# -----------------------------------------------------------------------------
# System options
# -----------------------------------------------------------------------------
[system]

# Enable or disable verbose startup and error output.
#
# When enabled, outputs more information to the LCD when starting up, and when
# MIDI/UART errors are detected.
#
# This also may hide the boot logo on smaller graphical displays.
#
# Values: on, off*
verbose = off

# Set the default synthesizer to be made active on startup.
#
# If the default synthesizer is unavailable (e.g. missing ROMs or SoundFonts),
# the first working synth is made active.
#
# Values: mt32*, soundfont
#
# mt32:      Use mt32emu (Munt) for Roland MT-32 emulation
# soundfont: Use FluidSynth for SoundFont synthesis
default_synth = soundfont

# Enable or disable support for USB devices.
#
# Disable this to speed up boot time if you are not using any USB devices.
#
[... 431 lines cut ...]
# of transport layer security/encryption. Therefore, you should NOT enable this
# feature on a public network or expose the Raspberry Pi to the Internet.
#
# The FTP server should be considered a convenience feature only, for performing
# updates and configuration changes without having to replace the SD card.
#
# Values: on, off*
ftp = on

# Set the FTP server username and password.
#
# Values: any ASCII string (*mt32-pi)
ftp_username = mt32-pi
ftp_password = p4ssw0rd!

```
</details>

## Videos/audio recordings

I can provide a short video later if needed.

## Additional information

n/a

dwhinham, 2022-03-03:
What's happening here is that disk I/O blocks everything else, and the incoming MIDI gets buffered up, then instantly dumped on the synth when disk I/O is done, i.e. you might get a few dozen Note-Ons all at once.

Should be an easy fix, I can just clear all the MIDI buffers after a SoundFont switch is done.

N.B. you shouldn't really be doing this. All the instruments and parameters are reset on switching and you'll be treated to an orchestra of pianos if you're hopping betweens synths and SoundFonts while a song is playing. This is by-design and won't be fixed.

c0d3h4x0r, 2022-03-03:
Understood about the otherwise problematic aspect of switching SoundFonts during playback.  But that, unlike the loud noise glitch, doesn't risk blowing out my ears or speakers.

Sent via Outlook for iOS<https://aka.ms/o0ukef>

commit 19b71b93 (2022-03-04): Ignore note-ons received during SoundFont switch

Fixes #247.
commit 28ba9207 (2022-03-04): Ignore note-ons received during SoundFont switch

Fixes #247.
commit 63a4a884 (2022-03-10): Ignore note-ons received during SoundFont switch

Fixes #247.
closed 2022-03-10 by dwhinham