dwhinham/mt32-pi #248: 💡 [FEATURE] Provide a per-SoundFont master gain override

c0d3h4x0r, 2022-03-02:
## Is your feature request related to a problem? Please describe.

Some SoundFonts are overall louder than others, such that it's impossible to set the master gain low enough to avoid clipping without other SoundFonts being too quiet.

## Describe the solution you'd like

Add a master gain override setting to each per-SoundFont configuration section, so the user can easily make quiet fonts louder or loud fonts quieter.

## Describe alternatives you've considered

While it's possible to hand-edit SoundFont files (using various apps) to reduce the volume for ever patch within the font, that's a lot of extra work just to get all the SoundFonts to be roughly the same volume.

Ideally, the user would never need to set the master gain level (for either Munt or FluidSynth) within the mt32-pi config.  Both sound engines ought to be smart enough to do the math to automatically set the gain to the loudest possible value that won't clip, based on information contained within each SoundFont combined with other settings (such as max polyphony, chorus settings, and reverb settings).  Clipping due to overgain should simply be made mathematically impossible.  However, I'm sure that's a lot more work, and probably requires changes to each of this respective sound engines, rather than to code maintained by this repo.

## Additional context

n/a

dwhinham, 2022-03-03:
Please test this build: https://github.com/dwhinham/mt32-pi/actions/runs/1929248605

It should now be possible to set `gain` in each FluidSynth profile section, and the main section contains the default as per the other effects settings.

It also contains an attempt at a fix for #247, so please check the behaviour when switching SoundFonts whilst music is playing.

c0d3h4x0r, 2022-03-03:
Awesome!  I'll give it a try tonight.

Sent via Outlook for iOS<https://aka.ms/o0ukef>

c0d3h4x0r, 2022-03-04:
Per-soundfont gain appears to work great for values greater than and less than 1.0.

#247 still repros, unfortunately -- no noticeable change.

Sent via Outlook for iOS<https://aka.ms/o0ukef>

c0d3h4x0r, 2022-03-04:
Hrm... actually, the per-soundfont gain override doesn't appear to stick.  It applies for a while, but then suddenly it switches back to the general gain level.

Sent via Outlook for iOS<https://aka.ms/o0ukef>

dwhinham, 2022-03-04:
Can you elaborate?

c0d3h4x0r, 2022-03-04:
I switch to a SoundFont that I've set to use gain = 0.5.  It loads.  I start playing Tyrian.  I notice the volume is greatly reduced.   Good.

A few minutes into playing the game, mid-song, the gain suddenly jumps back to full, without me doing anything.

dwhinham, 2022-03-04:
Reduced?

The initial gain is 0.2, which is the FluidSynth default. Setting it to 0.5 is an increase in volume.

It would be helpful if you could test this game with FluidSynth on a PC to understand the behaviour properly. Games are able to control the volume of individual MIDI channels, as well as MIDI master volume, and the gain setting is applied after all of those other volumes.

I'll try to find time to dig into this over the weekend...

c0d3h4x0r, 2022-03-04:
Ah crud, good point!  I forgot the global default isn't 1.0.  So now I've got quite the mystery on my hands, because that font definitely now sounds quieter when first selected.

I'll go back and retry with 0.075 or something instead.

dwhinham, 2022-03-04:
No problem.

Be sure that you are starting the soundtrack right from the beginning after switching SoundFonts. Mid-song is not a valid test, as any MIDI song is able to affect the volume in many ways independently of the gain and before gain is applied.

Example:

- You start the game.
- Song sets channel volumes to half.
- You switch SoundFont mid-song, the channel volumes are reset to full because switching SoundFonts is a full synth re-initialization.
- Reduced gain seems to have no effect, but in reality it does.

c0d3h4x0r, 2022-03-05:
Are you saying the per-soundfont gain overridable by MIDI rather than a true master mixing gain?

dwhinham, 2022-03-05:
No, not at all. It is a multiplier applied **after** MIDI channel volume and MIDI master volume.

I said above:
> Games are able to control the volume of individual MIDI channels, as well as note velocity and MIDI master volume, and the gain setting is applied after all of those other volumes.

c0d3h4x0r, 2022-03-09:
Got it.  Did more self-testing, and it looks good.  Thanks!

commit dc01c391 (2022-03-04): Add ability to override gain per-SoundFont

Closes #248.
commit 9602162a (2022-03-04): Add ability to override gain per-SoundFont

Closes #248.
commit a19f6c60 (2022-03-10): Add ability to override gain per-SoundFont

Closes #248.