dwhinham/mt32-pi #273: 💡 [FEATURE]

probonopd, 2022-04-15:
## Is your feature request related to a problem? Please describe.

The [Waveshare WM8960 audio hat](https://github.com/waveshare/WM8960-Audio-HAT/) is an inexpensive DAC with built in amplifier. 

## Describe the solution you'd like

It would be nice if it could be supported.

## Describe alternatives you've considered

Buying yet another DAC, just for mt32-pi. But the alternatives are often more costly and lack the amplifier.

## Additional context

```
pi@rpizero:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@rpizero:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

pi@rpizero:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- UU -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

probonopd, 2022-04-15:
Next step: set set `i2c_dac_address` in `mt32-pi.cfg` as described on https://github.com/dwhinham/mt32-pi/wiki/I²S-DACs and see what happens. Doing this now.

dwhinham, 2022-04-18:
This looks like a decent DAC, and as always I'd like to support it, but it's difficult for me to add support for new DACs without having the hardware in front of me.

I'll gladly accept a patch for it, on the understanding that I will be unable to test it or maintain it, otherwise I've added it to the Amazon wishlist, so if someone purchases it I'll try to get it working.

The approach I've used before is to do dumps of all of the i2c registers of the device, using Linux - once without the driver loaded (no devicetree etc), and once with it loaded (and in a working state that produces sound).

Then compare the two, cross-reference with the datasheet and try to work out the minimum required register pokes.

probonopd, 2022-04-19:
> The approach I've used before is to do dumps of all of the i2c registers of the device, using Linux

Using `i2cdump`? Do you have a command to do that?

dwhinham, 2022-04-19:
I can't remember, sorry; been a long time since I did this

probonopd, 2022-04-21:
A PR for supporting the WM8960 DAC has arrived in Circle:

https://github.com/rsta2/circle/pull/286

It even auto-detects well-known DACs on i2c.

dwhinham, 2022-04-21:
Yes, I can see - I get notifications.

I'll wait to see if it gets tested and/or amended by Rene before doing anything, because like I say I do not own this hardware, and I don't want any "auto-detection" to break existing setups.

commit 37c25673 (2022-06-13): Remove manual DAC init, use Circle auto-detection

Closes #273.
closed 2022-06-13 by dwhinham