dwhinham/mt32-pi #296: USB Serial Midi / RTPMidi passthrough to GPIO / USB / RTP midi TX

skadarnold, 2022-07-02:
More passthrough options between different protocols could be very useful. 

For example, the Vintage Serial / Raspberry Pi USB option is great, but it would also be useful to be able to passthrough the receive data to both the GPIO TX, RTPMidi respectively to be able to connect external synths. Or also from RTPMidi passthrough to GPIO / USB midi.

There might be other routing options to consider as well, but I think this could add some flexibility.

skadarnold, 2022-07-03:
As I am looking through the code for this, I might be able to help contribute to these enhancements if it is not currently being worked on elsewhere.

dwhinham, 2022-07-03:
Closing as:
- the issue template was ignored
- this is a duplicate of #138
- the feature is already being worked on

closed 2022-07-03 by dwhinham