dwhinham/mt32-pi #323: Continuous volume control with buttons💡 [FEATURE]

ruenes1, 2023-02-05:
## Is your feature request related to a problem? Please describe.

I use the MT32-Pi on a Pi zero 2 with a small hat with an oled screen and just buttons (no rotaries). 
With the 'simple_buttons' control scheme, two of the buttons control the volume, which is handy.
However, the button-based volume control seems to be discrete, that is, to go from 100 to 80, I have to press the 'minus vol' button 20 times. No matter how long the volume buttons are pressed, they only change one volume step per press.

## Describe the solution you'd like

It would be nice if the button-based volume control behaves similarly to a tv remote, that is, by continuously adjusting the volume while keeping the button pressed. It may even have some 'acceleration', that is, the longer the button is pressed, the faster the volume changes.

## Describe alternatives you've considered

I understand the current behaviour would not be a problem with a rotary-based volume control (1 volume step per encoder step), but I was not able to find any hats for the Pi zero that include a rotary encoder.

dwhinham, 2023-02-06:
Thank you for taking the time to complete the issue template properly and for the feature suggestion.

This was something I intended to implement some time ago but ended up forgetting about, as the current volume button code isn't very pleasant for the user.

I've implemented a hold action for the volume buttons which also includes the acceleration behaviour you suggested.

It works a little bit like a PC keyboard, where after a short delay the button will begin to repeat its action. The volume changes will accelerate from 10Hz to 50Hz over 3 seconds, which to me feels quite good.

Feel free to test [this build](https://github.com/dwhinham/mt32-pi/actions/runs/4107731989) and let me know what you think (simply download [kernels.zip](https://github.com/dwhinham/mt32-pi/suites/10813293074/artifacts/544073495) and replace the current kernels on your SD card).

If you like it, I'll include it in the next release.

ruenes1, 2023-02-06:
I've just tested it and I think it works great! Thanks!

By the way, I've noticed you reserved a pin for a 5th button (the rotary push button in the encoder scheme, but I understand a 5th standard button connected to that pin would also do the same, if support is added to the simple_buttons control scheme). 

I don't know if you have some function in mind for that button but, at least for MiSTer users, it could be handy to use it to send a 'reset hanging notes' command, which is needed quite often when a game crashes. In this way, all OSD functions could be mapped to physical buttons. If you think this makes sense, I can open a new feature request issue, if you want.

Best

dwhinham, 2023-02-06:
Thanks for testing the volume buttons, I'm glad it works well for you.

The reserved button will eventually be used for future menu navigation/selection. A "reset hanging notes" function could be one of dozens of functions accessible via the menu - to that end, I don't think it makes much sense to hard-code it to anything else in the meantime.

I haven't got a 100% final design in mind for the control schemes, but the reserved button will probably be used to show the menu and select items, with the volume buttons then temporarily becoming up/down buttons to move the cursor or change values.

Cheers

ruenes1, 2023-02-06:
Yes, what you have in mind makes more sense : )

commit 71fa6533 (2023-02-06): Implement volume button hold/repeat with delay

Improve the button handling code so that a pressed and held button will
now generate additional press events after a short delay.

An acceleration curve is also applied so that events are generated with
increasing frequency the longer the button is held down.

Closes #323.
closed 2023-02-23 by dwhinham