dwhinham/mt32-pi #327: 🐛 [BUG] Soundfont - SF2 file > 1gb will fail to load

guizmox, 2023-02-13:
## Hardware and software

- **mt32-pi version:** v0.12.1
- **Raspberry Pi model:** Raspberry Pi 4 CM4104000 module (4gb Lite, wifi) on a Serda McCake32 board.
- **USB devices:** None
- **MIDI host:** Miditemp Mulstistation MSX using Waveblaster port

## Bug description

Any attempt to load SF2 files bigger than 1gb will fail. "SF switch error". 
Those soundfonts successfully load using Fluidsynth for Windows (Fluidsynth 2.3.0).

## Steps to reproduce

1. Use a SF2 file that is bigger than 1gb
2. Use "change bank" to choose it
3. Wait a couple of seconds. "SF switch failed !"

## Expected behavior

File should load successfully : 3488 samples, 59231 instrument generators, 1611 preset generators

## Configuration file

<details>
<summary>Click to expand</summary>

```ini
#                                  ________    _______                      __
#                         __      /_____   `. /____   `.                   /__`
#     ____   ____     __/  /____  _______)  / ______)  / ____  ______      __
#   /   __ v  __  `./__   _____//_____    < /   _____. /____//   ___  `. /  /
#  /  /  /  /  /  /   /  /____  _______)  //  /______       /  /____/  //  .__
# /__/  /__/  /__/    \______//_________. /_________/      /   ______.  \____/
# /////////////////////////////////////////////////////// /  / //// /// // /
#                                                         ```
# mt32-pi.cfg: mt32-pi configuration file.
# Default options are marked with an asterisk (*).

# -----------------------------------------------------------------------------
# System options
# -----------------------------------------------------------------------------
[system]

# Enable or disable verbose startup and error output.
#
# When enabled, outputs more information to the LCD when starting up, and when
# MIDI/UART errors are detected.
#
# This also may hide the boot logo on smaller graphical displays.
#
# Values: on, off*
verbose = on

# Set the default synthesizer to be made active on startup.
#
# If the default synthesizer is unavailable (e.g. missing ROMs or SoundFonts),
# the first working synth is made active.
#
# Values: mt32*, soundfont
#
# mt32:      Use mt32emu (Munt) for Roland MT-32 emulation
# soundfont: Use FluidSynth for SoundFont synthesis
default_synth = soundfont

# Enable or disable support for USB devices.
#
# Disable this to speed up boot time if you are not using any USB devices.
#
# Values: on*, off
[... 423 lines cut ...]

# Set the network hostname.
#
# Values: a valid hostname using ASCII letters 'a' to 'z', digits 0-9, and
#         hyphens (mt32-pi*)
hostname = mt32-pi

# Enable or disable the RTP-MIDI/AppleMIDI server.
#
# This allows you to send MIDI data to mt32-pi over the network using macOS'
# built-in network MIDI features, or rtpMIDI by Tobias Erichsen on Windows.
#
# Values: on*, off
rtp_midi = off

```
</details>

## Videos/audio recordings

Nothing

## Additional information

I can provide the SF2 file if needed.

guizmox, 2023-02-14:
I used a HDMI out kernel version to check the LOG, and here's the result : 
It seems that the zoneallocator is fixed to 940 megabytes (line 4)
 
[image]

dwhinham, 2023-02-14:
Thank you for the detailed bug report and providing the logs, it's much appreciated.

The reason for this is because of the way Circle's memory map/allocation is implemented. Unfortunately, we don't get access to the whole 4GB (or 8GB on those models) at the moment.

On Pi 4, the first 1GB is called the "low heap", and Circle's internal memory allocator uses this. The next 2GB is the "high heap" and is handled by mt32-pi's custom allocator - this is what FluidSynth uses. The remaining 1GB on Pi 4/4GB (or 5GB on Pi4/8GB) is not usable right now.

That said, we are only getting 940MB of "high heap" according to your log, when it should be closer to 2GB, so this needs investigating.

(On Pi 2/3/02W there is no "high heap", only a "low heap". Here, 32MB is reserved for Circle's internal allocator, and the remainder is handled by ours).

Circle's internal memory allocator cannot keep track of large allocations, hence the need for a custom allocator, otherwise FluidSynth cannot unload SoundFonts and free the memory for when the user wants to load another one.

There are a few things that need to happen so that we can use the rest of the memory on Pi 4:
- Treat the "low heap" like we do on Pi 2/3/02W; reserve 32MB for the internal allocator, then give the remainder to a custom allocator so that FluidSynth can use it - this will give us (almost) an additional 1GB. This shouldn't be too difficult.
- Figure out why the custom allocator isn't getting the full 2GB that Circle should be able to provide from the upper heap.
- Investigate how we can extend the range of the upper heap so that the rest of the RAM can be used. The [reasons for the memory range being limited are given in the Circle documentation](https://github.com/rsta2/circle/blob/31cfd7303ce86266086dca0f17610d9b3f4067c4/doc/new-operator.txt#L39-L43) but it's not clear if we need to worry about the "specific DMA-handling" or not for this project.

I will need some time to figure out the best way to solve this.

P.S. thank you very much for the donation, I really appreciate it! :slightly_smiling_face:

guizmox, 2023-02-14:
Hello and thank you very much for the detailled answer, your project is really great, and useful for a lot of people including me.
That's seems to match with what I read from the "zoneallocator.cpp" file, I'm unfortunately not able to fork your code, I'm a C# dev. with no experience with C++
An additionnal 1gb would be great, I'm fully aware that bigger SF2 does not means better sound, but I'm not using MT32-pi as a GM/GS machine but more as a VST instrument to extend my sound palette :)

guizmox, 2023-02-15:
I read the memory.c Circle code, and this may explain things : 

[image]

In case of Pi4, Your custom allocator does the following : 
_**pMemorySystem->GetHeapFreeSpace(HEAP_HIGH);**_

According to Circle, using HEAP_ANY should also add the HEAP_LOW Free Space and return an addition of both.

I now have to figure out how I can compile your code :D

dwhinham, 2023-02-15:
It's not that simple. The custom allocator needs to make one large, single contiguous allocation from either one of those heaps, so using the size of "any" (the two heap sizes added together) doesn't make sense. Changing the `GetHeapFreeSpace()` call to use `HEAP_ANY` will give it a bigger number, but it will be incorrect.

The best solution (which I have wanted to do for a long time due to other related issues) will be to remove Circle's allocator completely and use the Zone allocator system-wide. There is a branch (`allocator`) where I have been working on this in the past, but there are some show-stopping bugs that need fixing before it can be used.

guizmox, 2023-02-15:
Interesting ! I'll look into that branch and see what I can find. Thank you :)

guizmox, 2023-02-16:
I was able to compile the code and make a slight change :

const size_t nHighHeapSize = pMemorySystem->GetHeapFreeSpace(HEAP_HIGH);
	if (nHighHeapSize)

The condition was always false (I can't figure out why). I did removed it and now my 1.5gb soundfont successfully loaded ! 
As you explained to me, the HIGH_HEAP zone was able to allocate 2048mb :) 

[image]