dwhinham/mt32-pi #330: 💡 [FEATURE]

gritd, 2023-02-23:
## Is your feature request related to a problem? Please describe.

_A clear and concise description of what problem(s) this feature request would solve. For example:_

I'm always frustrated when...

## Describe the solution you'd like

_A clear and concise description of what you want to happen._

## Describe alternatives you've considered

_A clear and concise description of any alternative solutions or features you've considered._

## Additional context

_Add any other context, videos, or audio recordings about the feature request here._

gritd, 2023-02-23:
I'm using a UDA1334 DAC board on mt32pi which works great, but when booting  the RPi starting up this DAC gives noise on the amp, would it be possible to just assign a GPIO port to output high once the system is ready and running so this artifact can be minimized? I dunno, let's say pin #26 which is near the WSEL for example.

There is a pin on the UDA1334 which -if set high- mutes the board, so this would be in reverse, but I guess puttting a pin high immediately after start (and set it to low when ready) of the RPi might be impossible ?

Anyway - I guess what I'm asking is a hardware pin status which indicates 'synth ready'

dwhinham, 2023-02-23:
At this time it's not reasonable to assign hard-coded GPIOs for random hardware quirks, because there are hundreds of RPi accessories that this could potentially conflict with, e.g. Pisound, which uses GPIOs to set sample rate.

Anything going into the code that messes with GPIOs needs some some of guarantee that it won't break something for some other user, so for now I will not be implementing this.

closed 2023-02-23 by dwhinham

icb-, 2023-02-23:
If your goal is to just assert the mute signal for a short time while the system boots, you can do this with something as simple as a resistor and capacitor. This is commonly used on microcontroller reset lines to hold them in reset while power stabilizes.

[Calculate the time constant](https://www.digikey.com/en/resources/conversion-calculators/conversion-calculator-time-constant) for how long you need it to stay muted, and pick values accordingly. For example, a 10µF capacitor and 100kΩ resistor would hold it in mute for about a second after power is applied.
[image]