dwhinham/mt32-pi #336: 🐛 [BUG] Some MIDI files play with wrong instruments in sound font mode

KyleJ61782, 2023-04-05:
## Hardware and software

- **mt32-pi version:** v0.13.1
- **Raspberry Pi models:** Zero 2 W and 3 A+
- **HATs:** mt32-pi-midi-hat for both 
- **USB devices:** None
- **MIDI host:** modern Windows 11 PC via Roland UM-ONE

## Bug description

When playing some music from the Descent using the sound font mode of the mt32-pi, some songs play with the wrong instruments. This happens with both mt32-pi devices that I have when connected to my Windows 11 machine with a Roland UM-ONE.

## Steps to reproduce

Play Game01.mid from Descent using any MIDI player that will output through the Roland UM-ONE.

## Expected behavior

I would expect it to play with the proper instruments.

## Configuration file

<details>
<summary>Click to expand</summary>

```ini
#                                  ________    _______                      __
#                         __      /_____   `. /____   `.                   /__`
#     ____   ____     __/  /____  _______)  / ______)  / ____  ______      __
#   /   __ v  __  `./__   _____//_____    < /   _____. /____//   ___  `. /  /
#  /  /  /  /  /  /   /  /____  _______)  //  /______       /  /____/  //  .__
# /__/  /__/  /__/    \______//_________. /_________/      /   ______.  \____/
# /////////////////////////////////////////////////////// /  / //// /// // /
#                                                         ```
# mt32-pi.cfg: mt32-pi configuration file.
# Default options are marked with an asterisk (*).

# -----------------------------------------------------------------------------
# System options
# -----------------------------------------------------------------------------
[system]

# Enable or disable verbose startup and error output.
#
# When enabled, outputs more information to the LCD when starting up, and when
# MIDI/UART errors are detected.
#
# This also may hide the boot logo on smaller graphical displays.
#
# Values: on, off*
verbose = on

# Set the default synthesizer to be made active on startup.
#
# If the default synthesizer is unavailable (e.g. missing ROMs or SoundFonts),
# the first working synth is made active.
#
# Values: mt32*, soundfont
#
# mt32:      Use mt32emu (Munt) for Roland MT-32 emulation
# soundfont: Use FluidSynth for SoundFont synthesis
default_synth = mt32

# Enable or disable support for USB devices.
#
# Disable this to speed up boot time if you are not using any USB devices.
#
# Values: on*, off
usb = off

[... 418 lines cut ...]

# Enable or disable the embedded FTP server.
#
# This FTP server is a very basic implementation which DOES NOT feature any kind
# of transport layer security/encryption. Therefore, you should NOT enable this
# feature on a public network or expose the Raspberry Pi to the Internet.
#
# The FTP server should be considered a convenience feature only, for performing
# updates and configuration changes without having to replace the SD card.
#
# Values: on, off*
ftp = on

# Set the FTP server username and password.
#
# Values: any ASCII string (mt32-pi*)
ftp_username = mt32-pi
ftp_password = REDACTED
```
</details>

## Videos/audio recordings

Here's how it sounds out of the mt32-pi: [Game01 - Broken.mp3](https://drive.google.com/file/d/1qvdbf3Sbu8uKLtGxuoOSmgwiymb09nZS/view?usp=share_link)
Here's how it sounds using Audacious, which uses AMIDI and Fluidsynth: [Game01 - Working.mp3](https://drive.google.com/file/d/19JbyR8QSbrQDHOjXwLAn7k82GFkPGl-J/view?usp=share_link)

dwhinham, 2023-04-05:
Can you please attach the MIDI file?

Note that Descent does not actually come with any .MID files as it uses its own custom format, so what you are trying to play is going to be some kind of unofficial "rip" of unknown quality.

Descent itself is known to be good with mt32-pi.

dwhinham, 2023-04-05:
(Sorry, mis-clicked the close button).

KyleJ61782, 2023-04-05:
That's a good point. Perhaps there's something pathological in the exported MIDI. Here's a link to the MIDI file:

[https://drive.google.com/file/d/16r9rQ5AvyxD2EgebLkTUYoDWrFZiEejh/view?usp=share_link](https://drive.google.com/file/d/16r9rQ5AvyxD2EgebLkTUYoDWrFZiEejh/view?usp=share_link)

I should note that the MIDI plays without issue on an SC-8820.

KyleJ61782, 2023-04-05:
I'm playing it using any player...I've used WinAmp, I've used Falcosoft's Midi Player as well. They're all outputting via the UM-ONE attached to my Windows 11 box. For Falcosoft's midi player, I've outputted both directly via the UM-ONE and also through the Windows midi mapper. Both give the same faulty result.

I'm having a hard time believing that messages aren't being sent properly since my SC-8820 is properly playing the midi file. It's almost like the initial messages are being sent too fast and the mt32-pi is inadvertently dropping them. Is there a way to log the incoming midi messages to a text file and then compare them to the midi file itself to see if there are missing messages?

KyleJ61782, 2023-04-05:
I should add that I'm a developer and willing to modify code and play around with things...I've got mt32-pi successfully building on my machine at the moment, so the turnaround time could be quick. I'm just unsure where to start looking.

dwhinham, 2023-04-05:
Ok. I'll check some of these programs with my Windows machine shortly, I spend 99% of my time in Linux so this could have fallen through the cracks in testing.

If I were to guess, the Windows MIDI stack or the UM-ONE's driver may be doing stuff to make the data stream more efficient before it goes over the wire, e.g. reordering the messages and using "Running Status", and there could be a bug in mt32-pi's parser.

If you want to help dig into it:

1. Start by enabling `verbose = on` in mt32-pi.cfg, as that will enable some errors on the LCD when the parser encounters unexpected stuff in the data stream.
2. The raw serial data is gathered here before it's sent to the parser. This is a good opportunity to log incoming bytes: https://github.com/dwhinham/mt32-pi/blob/ee645df6e38508e16e6a265fe89d13ce5c815b65/src/mt32pi.cpp#L933
3. They are then fed into the parser here: https://github.com/dwhinham/mt32-pi/blob/ee645df6e38508e16e6a265fe89d13ce5c815b65/src/midiparser.cpp#L36
2. When the parser sees a complete/valid message, it passes it to these callbacks, which then forward it to the current synth engine: https://github.com/dwhinham/mt32-pi/blob/main/src/mt32pi.cpp#L653-L683

dwhinham, 2023-04-05:
Just tested with TMIDI on Windows and now I'm able to reproduce the issue.

dwhinham, 2023-04-05:
[image]

This MIDI file seems corrupt to me. Above is a side by side log of events - on the left, a log dumped using [miditk-smf](https://github.com/SpotlightKid/miditk-smf). On the right, the raw data logged from mt32-pi.

The `B0` control change commands with parameters of 128 and 255 are bogus because values over 127 are not possible with MIDI. Data bytes are always between 0 and 127, so these are being misinterpreted or corrupt.

The Linux MIDI player tools must be silently ignoring these bad commands and just moving on to the next ones. But under Windows, TMIDI is outputting an `FF` byte.

`FF` is a "System Reset" command, which is supposed to reset the synth to power-up defaults, and that is exactly what mt32-pi is doing when it receives this message: https://github.com/dwhinham/mt32-pi/blob/ee645df6e38508e16e6a265fe89d13ce5c815b65/src/synth/soundfontsynth.cpp#L214-L220

The reason why it's "fine" on your real synthesizers is because they ignore this command. This is the MIDI implementation chart from the SC-8820, "X" means it does not handle this command:

[image]

Seeing as other Sound Canvases and the MU-series also seem to ignore this byte, there is an argument that mt32-pi should ignore it too to make the behaviour similar.

But the root cause is that this MIDI file is simply broken. I'd probably put the blame on the "HMP2MID.EXE" tool used to convert the Descent data mentioned in the MIDI's comments.

KyleJ61782, 2023-04-06:
Wowwww...well done. Thank you so much. I'm thinking this is just broken and that mt32-pi is acting within spec and arguably more correctly than the SC-8820 and related modules.

I apologize for taking your time and thank you for following up! I'll let you decide if you'd like to close this or actually consider this something that needs changing. 👍

KyleJ61782, 2023-04-06:
In case you were interested, I found another converter using a tool called wildmidi that allowed me to convert the HMP files and the result seems to work properly. Here's a Game01.mid file that appears to work:

[https://drive.google.com/file/d/1ex-US7orpTzI0xmAb_x9NYdq8DdOQ8Sk/view?usp=share_link](https://drive.google.com/file/d/1ex-US7orpTzI0xmAb_x9NYdq8DdOQ8Sk/view?usp=share_link)

Thank you so much for all of your help!

dwhinham, 2023-04-06:
No problem at all and no need to apologise, you were right to report this and I appreciate the time you took to test and provide all the information.

I'm pretty certain I've seen this behaviour in the past and was keen to get to the bottom of it.

It's great that you have found a better tool that produces a working MIDI conversion!

I was debating whether to make the changes to mt32-pi to remove the `FF` command support and I am leaning towards leaving it as-is and perhaps improving documentation and raising awareness that there are bad game MIDI rips out there.

If you agree with this, then yes, let's close this issue.

KyleJ61782, 2023-04-06:
Yep...I think it should be closed. Thanks for the help!

closed 2023-04-06 by KyleJ61782